thingsboard/thingsboard-gateway #762: [HELP] tb-gateway docker container - rest connector error

l33tkrew, 2022-03-03:
**Describe the issue**
I can connect my mqtt broker container with tb-gateway container without any problem. But, when i enabled the rest connector, the container restart itself continously. I attached the error logs and my rest configuration file.

**Configuration** (Attach your configuration file)
tbgateway.conf

```
thingsboard:
  host: 172.18.0.12
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: xxxxxxxxxxxxxxxxx
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
[... 304 lines cut ...]
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/composer.py", line 110, in compose_sequence_node
    while not self.check_event(SequenceEndEvent):
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python3.9/site-packages/PyYAML-6.0-py3.9-linux-x86_64.egg/yaml/parser.py", line 392, in parse_block_sequence_entry
    raise ParserError("while parsing a block collection", self.marks[-1],
yaml.parser.ParserError: while parsing a block collection
  in "//thingsboard_gateway/config/tb_gateway.yaml", line 35, column 3
expected <block end>, but found '<block sequence start>'
  in "//thingsboard_gateway/config/tb_gateway.yaml", line 80, column 4

Terminal will be reused by tasks, press any key to close it.
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version:  latest docker image
 - Python version: -

samson0v, 2022-03-04:
Hi @l33tkrew, please, send the command that you used to run the container or docker-compose file.

l33tkrew, 2022-03-04:
Part of my docker-compose file
```
  tb-gateway:
    restart: always
    image: "thingsboard/tb-gateway"
    container_name: gateway
    ports:
      - "5000:5000"
    volumes:
      - ./gateway/config:/thingsboard_gateway/config
      - ./gateway/logs:/thingsboard_gateway/logs
      - ./gateway/extensions:/thingsboard_gateway/extensions
```

I run docker-compose up -d to create non stop containers (restart: always). I manage the containers in VSCode Docker extension.  When I activate the REST connector, and restart container to apply changes. The container restarts itself again and again. Because the container give error logs after appying the changes. 
When I disabled the rest connector from tb_gateway.yaml, the container starts working.

l33tkrew, 2022-03-04:
Extra question:

When modifying the REST.connectory IP address. Should I paste the 127.0.0.1 or the container's own address?

samson0v, 2022-03-05:
@l33tkrew try to set `host: 0.0.0.0` and `expose` the port in docker-compose file.

l33tkrew, 2022-03-12:
I will set REST.connector's host address to 0.0.0.0 then I will expose 5000:5000 in docker-compose file. Is that correct?

samson0v, 2022-03-14:
@l33tkrew yes

closed 2022-03-17 by l33tkrew