thingsboard/thingsboard-gateway #763: [HELP]

redasdasd, 2022-03-03:
I have gateway version 2.9 installed on raspberry pi. Running with the following error.
--------------------------------------------------------------------------------------
pi@raspberrypi:~ $ systemctl status thingsboard-gateway.service
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-03-03 11:48:14 CET; 26min ago
 Main PID: 624 (python3)
    Tasks: 8 (limit: 3861)
   CGroup: /system.slice/thingsboard-gateway.service
           └─624 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Mar 03 12:14:31 raspberrypi python3[624]:     unit=unit_id)
Mar 03 12:14:31 raspberrypi python3[624]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
Mar 03 12:14:31 raspberrypi python3[624]:     return self.execute(request)
Mar 03 12:14:31 raspberrypi python3[624]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/pymodbus/client/sync.py", line 109, in execute
Mar 03 12:14:31 raspberrypi python3[624]:     return self.transaction.execute(request)
Mar 03 12:14:31 raspberrypi python3[624]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/pymodbus/transaction.py", line 201, in execute
Mar 03 12:14:31 raspberrypi python3[624]:     response, last_exception = self._retry_transaction(retries, "invalid", request, expected_response_length, full=full)
Mar 03 12:14:31 raspberrypi python3[624]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/pymodbus/transaction.py", line 250, in _retry_transaction
Mar 03 12:14:31 raspberrypi python3[624]:     in_waiting = self.client._in_waiting()
Mar 03 12:14:31 raspberrypi python3[624]: AttributeError: 'ModbusTcpClient' object has no attribute '_in_waiting'
------------------------------------------------------------------------------------------------------------------------------------------

samson0v, 2022-03-04:
Hi @redasdasd, install the latest version of the Gateway (3.0.1) and send the logs (DEBUG level), connector config file, Gateway config file. Also, describe the bug and steps to reproduce it.

samson0v, 2022-03-14:
Hi @redasdasd, any updates? Can we close the issue?