thingsboard/thingsboard-gateway #764: thingsboard gateway installation problem

debacs83, 2022-03-03:
Wanted to install thingsboard gateway following the instruction in:
https://thingsboard.io/docs/iot-gateway/getting-started/

It went well to some extent and have this configuration file under /etc/thingsboard-gateway/config/tb_gateway.yaml

```
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: MY_GW_ACCESS_TOKEN
...
...
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```

Change the Access Token with my gateway access token. 

Restarted it.

systemctl restart thingsboard-gateway

```
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
'thingsboard-gateway.service'を再起動するには認証が必要です。(it says that to restart thingsboard-gateway I need to authorize)
Authenticating as: kenkyu,,, (kenkyu)
Password:
==== AUTHENTICATION COMPLETE ===
```

But after that, when I used to check status, it gives the following error message.  

```
$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-03-03 20:03:51 JST; 33min ago
    Process: 178691 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
   Main PID: 178691 (code=exited, status=1/FAILURE)

 3月 03 20:03:51 spf python3[178691]:     create_key=_descriptor._internal_create_key,
 3月 03 20:03:51 spf python3[178691]: AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
 3月 03 20:03:51 spf systemd[1]: Started ThingsBoard Gateway.
 3月 03 20:03:51 spf systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
 3月 03 20:03:51 spf systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
 3月 03 20:03:51 spf systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
 3月 03 20:03:51 spf systemd[1]: Stopped ThingsBoard Gateway.
 3月 03 20:03:51 spf systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
 3月 03 20:03:51 spf systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
 3月 03 20:03:51 spf systemd[1]: Failed to start ThingsBoard Gateway.
```

Any help would be very much appreciated. 

 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard version 3.3.2
 - Thingsboard IoT Gateway version [3.0.1-1]
 - Python version[3.8]

debacs83, 2022-03-04:
Sorry, I made a mistake last time. I forgot to change the host from ' thingsboard.cloud' to 'localhost'. After editing this part, I restarted. But still the status remain same, with similar 'Failed to start ThingsBoard Gateway' message. 

Any other file to edit or check?

samson0v, 2022-03-04:
Hi @debacs83, please, send the whole tb_gateway.yaml file.

debacs83, 2022-03-04:
Hi @samson0v,
Thanks!  Here is my 'tb_gateway.yaml' file. Have tried with 'host: demo.thingsboard.io' with a different ACCESS_TOKEN, from my 'demo.thingsboard.io' account. But the result was same. Any help would be very much appreciated. 

```
thingsboard:
  #host: demo.thingsboard.io
  host: localhost
  port: 1883
  security:
    accessToken: MY_ACCESS_TOKEN
storage:
  type: memory
  read_records_count: 10
  max_records_count: 100
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json

#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

samson0v, 2022-03-05:
@debacs83, it is the actual config file, it seems that you are using the old version of the Gateway. Please, update it to the newest version.
```
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken:
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
  -
    name: REST Connector
    type: rest
    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json

```

debacs83, 2022-03-05:
Hi @samson0v,

Thank you so much! Seems it is working at last. Here is the message I got this time after 'restarting' and checking the 'status' with 'systemctl'. 

```
debasish@SonyVAIO:~/Downloads$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor prese>
     Active: active (running) since Sat 2022-03-05 19:34:22 JST; 33ms ago
   Main PID: 14102 (python3)
      Tasks: 1 (limit: 4549)
     Memory: 3.4M
     CGroup: /system.slice/thingsboard-gateway.service
             └─14102 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon;>

 3月 05 19:34:22 SonyVAIO systemd[1]: Started ThingsBoard Gateway.

```
A couple of things would like to confirm. 

1. In the 'tb_gateway.yaml' file, if I comment out everything, it is fine or if I make the mqtt-connector enable under 'connector', it is working. But if I enable the 'REST Connector' (as it is in your file), then the gateway is failed to start. Is it because only 'port: 1883' (mqtt) is being used? Or this 'port:1883' is for gateway to connect to TB server, nothing to do with 'connector'? 
2. This is related to gateway access token. Here we set the access token according the the gateway we create in ThingsBoard. But if we want to create more that one gateways then how could we set the access token? Is there any provision for setting more than one gateway in config file?

debacs83, 2022-03-05:
@samson0v 

Sorry to inform you that it has 'Stopped working'. I was trying to create Device under GW and sending the connect device msg. Expecting it will create Device under GW. But it was not working. When I tried to check the gateway 'status'. It shows the previous failed msg. I 'restarted'. Then the 'status' became 'running' again. But if I do the 'status' check twice consecutively it was showing the 'Failure' msg. Now, even after 'restart' the 'status' check back to 'Failure'. What am I doing wrong??    
I

debacs83, 2022-03-05:
Is this the reason? Am I restarting the gateway too soon? 

```
3月 05 21:23:18 SonyVAIO systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
 3月 05 21:23:18 SonyVAIO systemd[1]: Stopped ThingsBoard Gateway.
 3月 05 21:23:18 SonyVAIO systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
```

debacs83, 2022-03-06:
Update: 

Can create Device under GW by sending 'connect' message through Node-RED. It is automatically creating a device named 'Device A'. 
```
Topic: v1/gateway/connect
Message: {"device": "Device A"} 
```
But cannot send attributes to 'Device A' (temperature, humidity etc.)

samson0v, 2022-03-09:
@debacs83 send, please, your mqtt config file and I will show you what payload msg you have to send to Gateway.

samson0v, 2022-03-09:
@debacs83  For now, do you fix your issue with Gateway starting? If not, it seems you incorrectly install the Gateway.

debacs83, 2022-03-10:
Very sorry for the late reply. At home it was working partially (connecting device was ok, but not sending data as I wrote before). But could not repeat that at work. Ultimately could succeeded uninstalling and reinstalling IoT GW and could start it. Though at home I could at least connect the Devices to GW. But at work I could not even connect the Device to the Gateway through 'connect device' . Did the same thing as I reported 4 days back, using Node-RED to connect and send messages. Used GW access token in both 'tb_gateway.yaml' and in node-RED, security 'Username'. Password is kept blank.

I am sending the screen shots, both for 'Device connect' and sending information. It would be a great help if you please point out any problem. I was following these pages:
```
https://thingsboard.io/docs/iot-gateway/getting-started/
https://thingsboard.io/docs/reference/gateway-mqtt-api/
```

Screen shots:

[image]
[image]
[image]

[image]
[image]

Regards

debacs83, 2022-03-11:
{
  "broker": {
    "name": "Default Local Broker",
    "host": "127.0.0.1",
    "port": 1883,
    "clientId": "ThingsBoard_gateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "user",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          },
          {
            "type": "string",
            "key": "combine",
            "value": "${hum}:${temp}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
          "temperatureBytes": 2,
          "humidityBytes": 2,
          "batteryLevelBytes": 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeRequests": [
    {
      "retain": false,
      "topicFilter": "v1/devices/me/attributes/request",
      "deviceNameTopicExpression": "${SerialNumber}",
      "attributeNameJsonExpression": "${sensorModel}"
    }
  ],
  "attributeUpdates": [
    {
      "retain": true,
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}

debacs83, 2022-03-11:
In this 'mqtt.json' file the port is 1883. But in the screen-shot the port is set at 1882. They are different - one at home and one at work. So it is not incorrect.

samson0v, 2022-03-11:
Hi @debacs83, from your config you have to send the next msgs (example using mosquitto_pub):
`mosquitto_pub -h 127.0.0.1 -p 1883 -u user -p password -t /sensor/data -m '{"serialNumber": "YourDeviceName", "sensorType": "default", "hum": 12, "temp": 10, "sensorModel": "sdf"}'`

debacs83, 2022-03-12:
Thank you for your reply. I will try and let you know. What is the 'user' and 'password'? Usually password is device Access Token, but what should be the use? BTW, it seems I am sending temp & humidity values to a certain device. Where does thingsboard gateway comes to play in this message sending process?

samson0v, 2022-03-14:
@debacs83 user and password is your MQTT broker username and password and you have to specify them in the MQTT config file so that the paho client knows which credentials provide for the broker.
More info by the following links:
https://thingsboard.io/docs/iot-gateway/what-is-iot-gateway/
https://thingsboard.io/docs/iot-gateway/config/mqtt/

debacs83, 2022-03-14:
Thank you. I will let you know the result. Bogged down with another issue with TB with more urgent need to solve. 

I am not sure if this is related to gateway or not. Sorry if it is off the track. If so, please ignore it. 

Is it possible to send downlink message from Thingsboard to the source. And for this gateway is of any use or it should be something else?

debacs83, 2022-03-14:
BTW, I haven't installed other MQTT broker like 'mosquitto' or else. I am using Thingsboard's mqtt broker without separately installing any. What I knew that - 'ThingsBoard server nodes act as an MQTT Broker'.  At home I installed mosquitto and probably it was causing problem with the port (1883). I had to change the Thingsboard mqtt (bind_port) port to 1882 to make it working.  In that case shall I keep the 'user' and 'password' blank? Sorry for coming up with some probably naive questions.

samson0v, 2022-03-15:
> BTW, I haven't installed other MQTT broker like 'mosquitto' or else. I am using Thingsboard's mqtt broker without separately installing any. What I knew that - 'ThingsBoard server nodes act as an [...]

ThingsBoard works as a group of services, so you can use ThingsBoard MQTT Broker for Gateway using its address and port. 
I don't know if your MQTT Broker uses a username and password. If it is, you have to fill them into the config file.

debacs83, 2022-03-15:
Understood.

debacs83, 2022-03-22:
Thank you for all the helps. Yes, you can close this issue. If needed I will open a new one later one.

closed 2022-05-20 by imbeacon