thingsboard/thingsboard-gateway #765: [HELP] Mqtt Connector Problem

l33tkrew, 2022-03-03:
**Describe the issue**
I use latest tb-gateway docker image. But I cannot transfer data from tb-gateway to thingsboard through MQTT protocol

**Configuration** (Attach your configuration file)

tb_gateway.yaml
```
thingsboard:
  host: xxxxxxxxxxxxxxxxxx
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: xxxxxxxxxxxxxxxxxxxxxx
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
[... 258 lines cut ...]
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 3330, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/tb_client/tb_gateway_mqtt.py", line 76, in _on_message
    self._on_decoded_message(content, message)
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/tb_client/tb_gateway_mqtt.py", line 94, in _on_decoded_message
    target = content["device"] + "|*"
KeyError: 'device'

Terminal will be reused by tasks, press any key to close it.
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - Thingsboard IoT Gateway version: latest tb-gateway docker image
 - Python version

samson0v, 2022-03-04:
Hi @l33tkrew, please, send the logs with DEBUG level (in the logs.conf file change INFO to DEBUG).

l33tkrew, 2022-03-04:
Should I change all lines in the logs.conf file that start with "level=INFO" ?

samson0v, 2022-03-05:
@l33tkrew yes

samson0v, 2022-03-14:
@l33tkrew, any updates? Can we close the issue?

l33tkrew, 2022-03-18:
```

ame}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}"
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 154 - Mandatory key 'topicExpression' found in attributeUpdates handler: {"retain": true, "deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}"
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 154 - Mandatory key 'valueExpression' found in attributeUpdates handler: {"retain": true, "deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}"
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 162 - attributeUpdates handler has all mandatory keys => accepted: {"retain": true, "deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 165 - Number of accepted attributeUpdates handlers: 1"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 169 - Number of rejected attributeUpdates handlers: 0"
""2022-03-18 00:45:02" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 990 - Loaded devices:
 {'DEV-01': ['MQTT Broker Connector', '5']}"
""2022-03-18 00:45:02" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 611 - Send data Thread has been started successfully."
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 241 - MQTT Broker Connector connected to 172.20.0.42:11883 - successfully."
""2022-03-18 00:45:02" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 246 - Client <paho.mqtt.client.Client object at 0x7fe7aa8fb970>, userdata None, flags {'session present': 0}, extra_params ()"
""2022-03-18 00:45:02" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='JsonMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fe7a80542e0>, origin='/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py')"
""2022-03-18 00:45:02" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='JsonMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fe7a8054580>, origin='/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/mqtt/mqtt_uplink_converter.py')"
""2022-03-18 00:45:02" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='JsonMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fe7a8054730>, origin='/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py')"
""2022-03-18 00:45:02" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import JsonMqttUplinkConverter from /usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/mqtt."
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter JsonMqttUplinkConverter for topic /sensor/data - found!"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter JsonMqttUplinkConverter for topic /sensor/+/data - found!"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2022-03-18 00:45:02" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='CustomMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fe7a8054850>, origin='/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py')"
""2022-03-18 00:45:02" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import CustomMqttUplinkConverter from /usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions/mqtt."
""2022-03-18 00:45:02" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter CustomMqttUplinkConverter for topic /custom/sensors/+ - found!"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
""2022-03-18 00:45:02" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
[... cut ...]
""2022-03-18 00:45:20" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - run - 718 - {'deviceName': 'DEV-01', 'deviceType': '5', 'attributes': [{'model': 'abc'}, {'abc': 'on'}], 'telemetry': [{'temperature': '0.05'}, {'humidity': '50.02'}, {'combine': '50.02:0.05'}]}"
""2022-03-18 00:45:20" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _save_converted_msg - 352 - Successfully converted message from topic /sensor/data"
```

- I wrote "DEBUG" into .logs.conf file

- I can send only one mqtt telemetry until I restart the container. The same error can be seen above.

l33tkrew, 2022-04-07:
@samson0v any solution?

samson0v, 2022-04-08:
Hi @l33tkrew, as you can see from the logs (below screenshot)
[image]
You send the attribute request with the data: `"{'DEV-01': {'model': 'abc', 'abc': 'on'}}""`, but in your configuration file, you have the next:
```
"deviceNameTopicExpression": "${SerialNumber}",
"attributeNameJsonExpression": "${sensorModel}"
```
So Gateway can't parse needed parameters from the data.
In the right way, your data have to be the next one: `{"SerialNumber":  "my_device", "sensorModel": "mySharedAttribute"}`

l33tkrew, 2022-04-08:
I think there is some bugs in tb-gateway docker version. Because, on my local computer, I built docker image from the latest github repository (main), I works well. I send exactly the same json data. 

This is the command that I run to test
```
mosquitto_pub -h "xxxxxxx" -p "1883" -t "/sensor/data" -m '{"serialNumber": "tempDev", "sensorType": "Thermometer", "sensorModel": "T1000", "temp": 42, "hum": 58}'
```

This is the mqtt.json file on my local computer (It is working)

```
{
  "broker": {
    "name": "Default Local Broker",
    "host": "..........",
    "port": 1883,
    "clientId": "tb_gateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "test",
      "password": "test"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          },
          {
            "type": "string",
            "key": "combine",
            "value": "${hum}:${temp}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
          "temperatureBytes": 2,
          "humidityBytes": 2,
          "batteryLevelBytes": 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeRequests": [
    {
      "retain": false,
      "topicFilter": "v1/devices/me/attributes/request",
      "deviceNameTopicExpression": "${SerialNumber}",
      "attributeNameJsonExpression": "${sensorModel}"
    }
  ],
  "attributeUpdates": [
    {
      "retain": true,
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
```

There are the same lines that you wrote above. However, it works well in my container with my docker image. 
I think there is a problem in official docker image (on docker hub). It seems Official Docker image should be updated.

l33tkrew, 2022-04-08:
Here is the partial log of my container which run via my local image -not the official image-

```
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 1"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 4"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/+/connect, subscription message id = 5"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/disconnect, subscription message id = 6"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-04-08 20:35:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subscription message id = 7"
""2022-04-08 20:35:34" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - run - 718 - {'deviceName': 'tempDev', 'deviceType': 'test', 'attributes': [{'model': 'sada'}, {'sada': 'on'}], 'telemetry': [{'temperature': '20.03'}, {'combine': 'None:20.03'}]}"
""2022-04-08 20:35:34" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _save_converted_msg - 352 - Successfully converted message from topic /sensor/data"
""2022-04-08 20:35:34" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attribute_update_callback - 909 - Attribute request received with content: "{'tempDev': {'model': 'sada', 'sada': 'on'}}""
""2022-04-08 20:35:34" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attribute_update_callback - 910 - ()"
""2022-04-08 20:35:34" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 306 - Received data: {'tempDev': {'model': 'sada', 'sada': 'on'}}"
""2022-04-08 20:35:39" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - run - 718 - {'deviceName': 'tempDev', 'deviceType': 'test', 'attributes': [{'model': 'sada'}, {'sada': 'on'}], 'telemetry': [{'temperature': '20.02'}, {'combine': 'None:20.02'}]}"
""2022-04-08 20:35:39" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _save_converted_msg - 352 - Successfully converted message from topic /sensor/data"
""2022-04-08 20:35:39" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attribute_update_callback - 909 - Attribute request received with content: "{'tempDev': {'model': 'sada', 'sada': 'on'}}""
""2022-04-08 20:35:39" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attribute_update_callback - 910 - ()"
""2022-04-08 20:35:39" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 306 - Received data: {'tempDev': {'model': 'sada', 'sada': 'on'}}"
```

mqtt.json (works well)

```
{
  "broker": {
    "name": "Default Local Broker",
    "host": "***********",
    "port": 1883,
    "clientId": "tb_gateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "test",
      "password": "test"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
[... 211 lines cut ...]
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json

```

samson0v, 2022-08-04:
Hi @l33tkrew, update your docker container to the newest version and let us know if the bug is gone.

samson0v, 2022-08-16:
Hi @l33tkrew, any updates? Can we close the issue?

closed 2022-10-13 by imbeacon