thingsboard/thingsboard-gateway #766: [HELP] Telemetry Split REST connector

DjFoxzi, 2022-03-04:
**Describe the issue**
Hello, I have a question regarding the converter of the rest connector, when I send a telemetry with ts, it assigns bn the ts to each field but sends each of them individually to the thingsboard, as I am processing it together in the rules engine I need that are sent together in the values, and not each one in a separate value

**Configuration** 
[rest.zip](https://github.com/thingsboard/thingsboard-gateway/files/8188694/rest.zip)

**REST Converter :**

json: {"serial": 20,"ts":1638081400,"idDev": 1,"lonDev": -2.5029433,"latDev":af": " CTO 4 ","meter": "Nuevo","idmeter":  1}

Result
CONVERTED_DATA: {'deviceName': 'optimus 20', 'deviceType': 'optimus', 'attributes': [], 'telemetry': [{'ts': 1638081400, 'values': {'idDev': '1'}}, {'ts': 1638081400, 'values': {'lonDev': '-2.5029433'}}, {'ts': 1638081400, 'values': {'latDev': '2.6171326'}}, {'ts': 1638081400, 'values': {'idTraf': '1'}}, {'ts': 1638081400, 'values': {'idmeter': '1'}}, {'ts': 1638081400, 'values': {'ubTraf': ' CTO 4 '}}, {'ts': 1638081400, 'values': {'meter': 'Nuevo'}}]}"

**Versions :**
 - Thingsboard IoT Gateway version [e.g. 3.0.1] (Docker)
 - Python version[ 3.9]

samson0v, 2022-03-05:
Hi @DjFoxzi, it seems that it is a bug in the converter, we will investigate and let you know.
Thanks for your interest in ThingsBoard IoT Gateway!

DjFoxzi, 2022-03-05:
It's ok I solved it by modifying the ts from the thingsboard for now, at the moment I have another question, can I use the format [{telemetry},{telemetry},...] (List of Json Telemetry Values) to send telemetry through REST?

samson0v, 2022-03-09:
@DjFoxzi yes, you can

samson0v, 2022-03-14:
@DjFoxzi, can we close the issue?

closed 2022-03-24 by DjFoxzi