thingsboard/thingsboard-gateway #768: [BUG] Modbus configuration JSON file: ERROR | list indices must be integers or slices, not dict

alexcorvis84, 2022-03-09:
**Describe the bug**
I have a modbus_serial.json configuration file which contains few attributes to read from Modbus. 
If the key "attrributes" array has **more than 25 elements** I'm getting the following error:
```
""2022-03-09 13:10:29" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"

```
However, if  key "attrributes"  are **less or equals to 25 elements**, modbus connector is working ok.

So it makes me wonder if:  Is the "attributes" key array limited to just 25 elements? Or is it a bug?

**Connector name (If bug in the some connector):**
Modbus Connector

**Error traceback (If available):**
```
""2022-03-09 12:24:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
""2022-03-09 12:24:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
""2022-03-09 12:24:01" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import ModbusConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus."
""2022-03-09 12:24:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
""2022-03-09 12:24:02" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __rpc_gateway_processing - 771 - Received RPC request to the gateway, id: 16, method: gateway_device_deleted"
""2022-03-09 12:24:02" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __rpc_gateway_processing - 779 - Remote shell is disabled."
""2022-03-09 12:24:02" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __process_deleted_gateway_devices - 329 - Received deleted gateway device notification: CPD"
""2022-03-09 12:24:03" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"
""2022-03-09 12:24:08" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"
""2022-03-09 12:24:13" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"
```

**Versions (please complete the following information):**
 - OS: [Raspbian GNU/Linux 10 (buster)]
 - ThingsBoard IoT gateway version: 3.0.1
 - Python 3.7.3

**Modbus Connector configuration file example (modbus_serial.json)**
```
{
    "master":{
       "slaves":[
          {
             "name":"Modbus Default Server",
             "type":"serial",
             "method":"rtu",
             "port":"/dev/ttyUSB0",
             "baudrate":19200,
             "timeout":35,
             "byteOrder":"BIG",
             "wordOrder":"BIG",
             "retries":true,
             "retryOnEmpty":true,
             "retryOnInvalid":true,
             "pollPeriod":5000,
             "unitId":3,
             "deviceName":"CPD",
             "deviceType":"Analizador",
             "attributesPollPeriod":5000,
             "timeseriesPollPeriod":5000,
             "sendDataOnlyOnChange":false,
             "attributes":[
                {
                    "byteOrder":"BIG",
                    "tag":"Corriente fase III",
                    "type":"32int",
                    "functionCode":3,
                    "registerCount":2,
                    "address":22
                 },
                {
                   "byteOrder":"BIG",
                   "tag":"Corriente fase II",
                   "type":"32int",
[... 178 lines cut ...]
                   "address":78
                },
                {
                   "byteOrder":"BIG",
                   "tag":"Potencia aparente fase II",
                   "type":"32int",
                   "functionCode":3,
                   "registerCount":2,
                   "address":76
                },
                {
                   "byteOrder":"BIG",
                   "tag":"Potencia reactiva fase II",
                   "type":"32int",
                   "functionCode":3,
                   "registerCount":2,
                   "address":16
                }
             ]
          }
       ]
    },
    "slave":null
}
```

samson0v, 2022-03-09:
Hi @alexcorvis84, we will investigate your problem and let you know about the results.
Thanks for your interest in ThingsBoard IoT Gateway!

alexcorvis84, 2022-03-09:
> Hi @alexcorvis84, we will investigate your problem and let you know about the results. Thanks for your interest in ThingsBoard IoT Gateway!

Thank you @samson0v . I'll stay tuned ;-)

Meanwhile, FYI: we've found a workaroud that seems to be working.

Looking at the error, seems that it's produced due in [tb_gateway_service.py](https://github.com/thingsboard/thingsboard-gateway/blob/d8f21161673a2532f20a2b79a38257ee6da34abc/thingsboard_gateway/gateway/tb_gateway_service.py#L563) value **"maxPayloadSizeBytes"** is setup at 1024 by **default**. 

I tried to add the key **maxPayloadSizeBytes** in my **tb_gateway.yaml** configuration as below (increasing number up to 8192 bytes):

```
thingsboard:
  maxPayloadSizeBytes: 8192
  ...
```

And now, my **modbus_serial.json** conector configuration file which contains more than 25 "attributes" elements (actually 35), is working without any problem 🎉.

[image]

Hope you'll find interesting in order to update and improve documentation if that's the case.

BTW... Stay Safe!  🇺🇦💪

closed 2022-03-10 by imbeacon