thingsboard/thingsboard-gateway #769: [HELP] Missing MQTT elements are returned to thingsboard as variable strings

KrakenIsHere, 2022-03-10:
**The issue**
I have several values coming into my MQTT broker, some values come more often than others.
While using the MQTT connector, what I get is that when one of those less common values isn't in the published message, the reported value in thingsboard shows as the string value like this "${valueToGet}" instead of ignoring it.

[image]

**Configuration**
MQTT.json

{
  "broker": {
    "name": "Default Local Broker",
    "host": "127.0.0.1",
    "port": 1883,
    "clientId": "ThingsBoard_gateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "",
      "password": ""
    }
  },
  "mapping": [
    {
      "topicFilter": "energy_meter_electrical_main/measurements",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "IOT2056_M${main}_${phases}p",
        "deviceTypeJsonExpression": "EnergyMeter_Electrical", 
        "timeout": 60000,
        "attributes": [
          {
            "type": "long",
            "key": "phases",
            "value": "${phases}"
          }, 
          {
            "type": "long",
            "key": "main",
            "value": "${main}"
          },
          {
            "type": "string",
            "key": "module",
            "value": "${module}"
          }, 
          {
            "type": "string",
            "key": "version",
            "value": "${version}"
          }, 
          {
            "type": "string",
            "key": "make",
            "value": "${make}"
          }, 
          {
            "type": "string",
            "key": "model",
            "value": "${model}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "active_power_positive",
            "value": "${active_power_positive}"
          },
          {
            "type": "double",
            "key": "active_power_negative",
            "value": "${active_power_negative}"
          },
          {
            "type": "double",
            "key": "reactive_power_positive",
            "value": "${reactive_power_positive}"
          },
          {
            "type": "double",
            "key": "reactive_power_negative",
            "value": "${reactive_power_negative}"
          },
          {
            "type": "double",
            "key": "apparent_power_positive",
            "value": "${apparent_power_positive}"
          },
          {
            "type": "double",
            "key": "apparent_power_negative",
            "value": "${apparent_power_negative}"
          },
          {
            "type": "double",
            "key": "power_factor",
            "value": "${power_factor}"
          },
          {
            "type": "double",
            "key": "active_energy_import",
            "value": "${active_energy_import}"
          },
          {
            "type": "double",
            "key": "active_energy_export",
            "value": "${active_energy_export}"
          },
          {
            "type": "double",
            "key": "reactive_energy_import",
            "value": "${reactive_energy_import}"
          },
          {
            "type": "double",
            "key": "reactive_energy_export",
            "value": "${reactive_energy_export}"
          }, 
          {
            "type": "double",
            "key": "apparent_energy_import",
            "value": "${apparent_energy_import}"
          },
          {
            "type": "double",
            "key": "apparent_energy_export",
            "value": "${apparent_energy_export}"
          }
        ]
      }
    },
    {
      "topicFilter": "energy_meter_electrical_circuit/measurements",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "IOT2056_M${main}_C${circuit}_${phases}p",
        "deviceTypeJsonExpression": "EnergyMeter_Electrical",
        "timeout": 60000,
        "attributes": [
          {
            "type": "long",
            "key": "phases",
            "value": "${phases}"
          }, 
          {
            "type": "long",
            "key": "main",
            "value": "${main}"
          }, 
          {
            "type": "long",
            "key": "circuit",
            "value": "${circuit}"
          },
          {
            "type": "string",
            "key": "module",
            "value": "${module}"
          }, 
          {
            "type": "string",
            "key": "version",
            "value": "${version}"
          }, 
          {
            "type": "string",
            "key": "make",
            "value": "${make}"
          }, 
          {
            "type": "string",
            "key": "model",
            "value": "${model}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "active_power_positive",
            "value": "${active_power_positive}"
          },
          {
            "type": "double",
            "key": "active_power_negative",
            "value": "${active_power_negative}"
          },
          {
            "type": "double",
            "key": "power_factor",
            "value": "${power_factor}"
          },
          {
            "type": "double",
            "key": "active_energy_import",
            "value": "${active_energy_import}"
          },
          {
            "type": "double",
            "key": "active_energy_export",
            "value": "${active_energy_export}"
          }
        ]
      }
    }
  ],
  "connectRequests": [],
  "disconnectRequests": [],
  "attributeRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": []
}

**Versions:**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

rduivenvoorde, 2022-03-10:
Yeah, I've noticed that too. You think that is an issue? Or not? (For me it was not earlier that I found out that I was actually missing those fields/attributes in the json that I found out that THAT was the issue :-) ).

Now I/you know, you could argue about what is the issue here: the publisher of the json-message which sents half/wrong message, or the gateway who publishes the variable/template string.

If you say: ignore it, you would not see that the json that is being sent is wrong, at least now you see it?
Would it not be better if the mqtt publisher would include some (empty) string in the messages?

Not sure what others think...

KrakenIsHere, 2022-03-10:
As I saw it in the Java version of the gateway, it would appear to simply ignore them, and not include them in the message sent to thingsboard. This makes sense to me and has become the behavior I'd expect.

That said, I think that if it would just send an empty string or a null value, that could be workable for my use case.

samson0v, 2022-03-14:
Hi @KrakenIsHere, we will add a new parameter to the config file.
Thanks for your interest in ThingsBoard IoT Gateway!

closed 2022-03-14 by imbeacon