thingsboard/thingsboard-gateway #770: [BUG] Gateway docker image is not loading mqtt extensions (wrong internal pythonpath?)

rduivenvoorde, 2022-03-10:
**Describe the bug**

In a setup running containers I fail to use a custom mqtt python extension in the mounted  `/thingsboard_gateway/extensions` directory.

The gateway only 'finds' the custom python code when I copy  it IN the running container in:
`/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions/mqtt`

To reproduce, I went back to a minimal setup by just using a official docker image on a laptop and copy the custom extension in it:

1) run vanilla official image:

`docker run -it -v ~/.tb-gateway/logs:/thingsboard_gateway/logs -v ~/.tb-gateway/extensions:/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/thingsboard_gateway/config --name tb-gateway thingsboard/tb-gateway`

2) bring down, edit mqtt.json so it connects to an actual local broker, edit tb_gateway.yaml to connect to thingsboard.cloud for example

3) copy the available default `~/.tb-gateway/extensions:/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py` and create a `custom_mqtt_uplink_converter2.py` next to it (also edit the classname in it from `CustomMqttUplinkConverter` to `CustomMqttUplinkConverter2`

4) now edit the mqtt.json in `~/.tb-gateway/config:/thingsboard_gateway/config` to include a new topic, copy the custom/sensors one so it should use the new CustomMqttUplinkConverter2 class

```
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
          "temperatureBytes": 2,
          "humidityBytes": 2,
          "batteryLevelBytes": 1
        }
      }
    },
    {
      "topicFilter": "/custom/sensors2/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter2",
        "extension-config": {
          "temperatureBytes": 2,
          "humidityBytes": 2,
          "batteryLevelBytes": 1
        }
      }
    }

```

Now fire up the container again, and note the following in the output:
`""2022-03-10 11:35:58" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 269 - Cannot find converter for /custom/sensors2/+ topic"`

I think I went pretty far in debugging this (adding some tools to an images, editing the python files to see some output, see https://github.com/thingsboard/thingsboard-gateway/issues/750), and I think the crux is that the pythonpath in the container is NOT picking up the python code in the `/thingsboard_gateway/extensions` directory.

By going into the container and trying to call/see `TBModuleLoader.find_paths()` and `TBModuleLoader.import_module(...)` I see:

```
[
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors'
]

[... 60 lines cut ...]
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
""2022-03-10 11:35:58" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 269 - Cannot find converter for /custom/sensors2/+ topic"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 1"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 4"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/+/connect, subscription message id = 5"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/disconnect, subscription message id = 6"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
""2022-03-10 11:35:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 333 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subscription message id = 7"
""2022-03-10 11:35:59" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
""2022-03-10 11:35:59" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 2 for device *"
""2022-03-10 11:37:27" - |ERROR| - [custom_mqtt_uplink_converter.py] - custom_mqtt_uplink_converter - convert - 45 - Error in converter, for config: 

```

**Versions (please complete the following information):**

 - Thingsboard IoT Gateway version 3.0.1 official docker images

LennardWesterveld, 2022-03-11:
Thats because it runs the setup.py inside the docker build. That is also installing thingsboard-gateway it self inside the site-packages directory. If you should replace the setup.py with `pip install -e .` then the thingsboard-gateway directory will stay where it is placed.

rduivenvoorde, 2022-03-11:
@LennardWesterveld mmm, ok, so then is this considered an issue in the container run? 

Then I see 2 options:

- The gateway SHOULD be installed in the containers python, but then the pythonpath used by the internals should be fixed to pickup the /thingsboard_gateway/extensions

- The gateway should be ran from the . dir, without installing. Not sure what the 'setup.py' is adding? Also not sure if the extensions dir is picked up then...

Anybody can give info about what would be the best option here?

LennardWesterveld, 2022-03-14:
@rduivenvoorde this would help you I guess ;) https://gist.github.com/LennardWesterveld/bb8c2c95c2f6649c1306c56478caebd2 this is my docker file for now. In my opinion thingsboard-gateway, docker file itself should also move towards this setup.

samson0v, 2022-04-13:
Hi @rduivenvoorde, you can try to use this [solution](https://github.com/thingsboard/thingsboard-gateway/issues/615).

rduivenvoorde, 2022-04-13:
@samson0v yes, thanks that is what I do now. 
In that issue, you say: "Hi ilseva, we fixed this bug so in the next release you can use docker image without extra configurations."
That was nov 2021, did that already land in a (docker)-release?

samson0v, 2022-04-22:
Hi @rduivenvoorde it partially solved the problem but as you can see there are some exceptions.
Will fix.

closed 2022-08-08 by imbeacon

raomin, 2022-08-21:
Hi, I also ran into this issue and I would suggest to keep this issue open until it is fixed. ie docker image users can use the `~/.tb-gateway/extensions` folder to put in their custom converter.