thingsboard/thingsboard-gateway #773: [HELP] I cannot run tb-gateway.py file

l33tkrew, 2022-03-13:
**Describe the issue**
I installed "git clone https://github.com/thingsboard/thingsboard-gateway.git" the latest version through github. But I encounter with errors when I run "python3 tb_gateway.py"

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

```
thingsboard:
  host: xxxxxxxxxxxxxxxxxx
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: xxxxxxxxxxxxxxxxxxxxxxxx
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json
```

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
Traceback (most recent call last):
  File "tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 45, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 24, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkConverter
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_downlink_converter.py", line 21, in <module>
    from thingsboard_gateway.gateway.proto.messages_pb2 import *
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/gateway/proto/messages_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2022-03-14:
Hi @l33tkrew, try the next command `pip install --upgrade protobuf`

igloodan, 2022-03-15:
@samson0v 

That worked for me when I had the same problem.

samson0v, 2022-03-16:
@l33tkrew any updates? Can we close the issue?

closed 2022-03-17 by l33tkrew