thingsboard/thingsboard-gateway #775: [HELP] Not writing float attributes accordingly through Modbus TCP

mcavalcante, 2022-03-14:
**Describe the issue**
I´ve configured a ThingsBoard gateway as instructed in the link https://thingsboard.io/docs/iot-gateway/getting-started/ (Ubuntu), where I connected a PLC (model PLC-300 / WEG) programmed to receive some parameters through Modbus TCP. I could successfuly read and write integer values to this PLC, but for float it is simply not working for writing (ISSUE A), but reading (ISSUE B) I´d like to check with you if the behaviour is Ok.

The problem is that the numbers I write in the attributes are been written wrongly. For example: 

The configuration for the Modbus connector is as following (/etc/thingsboard-gateway/config/modbus.json):
For reading (seems to be working)
        "attributes": [
         {
            "tag": "var_a",
            "type": "32float",
            "wordOrder": "LITTLE",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8016
         },

For writing (not working Ok)
        "attributeUpdates":[
...
         {
            "tag": "svar_a",
            "type": "32float",
            "functionCode": 16,
            "objectsCount": 2,
            "address": 8016
         },

TO REPRODUCE THE ISSUES:
ISSUE A (WRITE)
1) Iin ThingsBoard UI, I click on "Devices" on vertical menu;
2) Find and click on device "PLC-300", and in the new windows opened, click on "Attributes";
3) Select "Shared Attributes" and create one for the key "svar_a', type '123 double', value "6.1" and click "Add";
4) On the PLC, modbus address 8016, now I see the value "1.8367E-41" (it should be 6,1).

ISSUE B (READ)
1) In the PLC side, throught its HMI, I set a variable called VAR_A (the same one that I tried) to "6.1";
2) In ThingsBoard, I navigate throught Devices, "PLC-300", Attributes, Client Attributes;
3) I see that the "var_a" attribute, the same one set on the step 1, is with the value "6.099999904632568";
Q: For the issue B, should I consider it a problem, or is it expected?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2022-03-15:
Hi @mcavalcante, thanks for your interest in ThingsBoard IoT Gateway! We will investigate this problem.

mcavalcante, 2022-03-16:
> Hi @mcavalcante, thanks for your interest in ThingsBoard IoT Gateway! We will investigate this problem.

After doing some researches, I found a way to put it working, but I´m not sure it is the correct way. Firstly, I´ve set the object "svar_a" like this:
```
         {
            "tag": "svar_a",
            "type": "32float",
            "byteOrder": "BIG",
            "wordOrder": "LITTLE",
            "functionCode": 16,
            "objectsCount": 2,
            "address": 8016
         },
```
... and, in the file /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py, I´ve changed some lines of the method __process_rpc_request in the following way:
```
                try:
                    #rpc_command_config[PAYLOAD_PARAMETER] = converted_data[0]
                    rpc_command_config[PAYLOAD_PARAMETER] = converted_data
                except IndexError:
                    rpc_command_config[PAYLOAD_PARAMETER] = converted_data
```
For some reason, when the variable PAYLOAD_PARAMETER has some value ("payload" in my personal scenario), it was only building the modbus frame with only one object (address 8016). Now it is writing both ones, the 8016 and 8017.

Would it be a bug or some misconfiguration that I did?

samson0v, 2022-07-18:
Hi @mcavalcante, please, update your Gateway to the newest version via the master branch and let us know if it helps.

closed 2022-11-01 by imbeacon

memind-rad, 2023-03-17:
Hello @samson0v,

Sorry, we´ve retaken this part of the project few weeks ago and I haddn´t see your message. I´ve tried with the version below and I guess we had an step back. Now, even with the work around suggested here in march 16º 2022, I can´t write float variables from TB toward the the PLC (ISSUE A mentioned above). However, for reading the float vars, it´s Ok.

`""2023-03-17 10:35:13" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"`

Work around:
```
                try:
                    i = 0
                    for cd in converted_data:
                        if (cd > 0):
                            i += 1

                    if (i >= 2):
                        rpc_command_config[PAYLOAD_PARAMETER] = converted_data
                    else:
                        rpc_command_config[PAYLOAD_PARAMETER] = converted_data[0]
                except IndexError:
                    rpc_command_config[PAYLOAD_PARAMETER] = converted_data
            try:
                response = self.__function_to_device(device, rpc_command_config)
```