thingsboard/thingsboard-gateway #776: [HELP] IoT Gateway: Is there any possibility to send configuration updates remotely?

alexcorvis84, 2022-03-15:
**Describe the issue**

Is there any possibility to send configuration updates to IoT Gateway from Thingsboard platform? As far as I've checked on the [online documentation](https://thingsboard.io/docs/reference/gateway-mqtt-api/) it mentions that MQTT device API is able to **receive configuration updates**, yet there's no example listed how this feature can be performed 😟

_For example:_ If i want to update the **tb_gateway.yaml** or any of the configuration files (*.json) from connectors (in /etc/thingsboard-gateway/config) remotely from Thingsboard platform, is this feature available?

If not, what would be the procedure to be able to edit these files remotely? I tried to create an RPC remote shell, and tried to modify some of them via command, but seems it can be performed (see screenshot below).

[image]

Also, I've noticed that TAB autocomple command doesn't work via RPC remote shell, which makes navigation a bit tricky. Do you plan to add this feature? I think it is quite important to ease remote shell management.

Many thanks in advance for your support and opensource development 😉

**Versions (please complete the following information):**
 - OS: Raspbian GNU/Linux 10 (buster)]
 - ThingsBoard IoT gateway version: 3.0.1
 - Python 3.7.3

samson0v, 2022-03-15:
Hi @alexcorvis84, yes, it is possible.
We have a group of Gateway widgets of the remote configuration. You have to create a dashboard or use created one, click on the "create new widget" button, scroll a little bit down and finally, you will see the "Gateway widgets" group.
[image]

alexcorvis84, 2022-03-15:
> Hi @alexcorvis84, yes, it is possible. We have a group of Gateway widgets of the remote configuration. You have to create a dashboard or use created one, click on the "create new widget" button, scroll a little bit down and finally, you will see the "Gateway widgets" group. [image]
> 
> Thanks for your interest in ThingsBoard IoT Gateway!

My bad! I didn't check that option 😅! 

What about to include TAB autocomple command via RPC remote shell? Would it be possible to include?

Thanks a lot @samson0v for your quick reply!

samson0v, 2022-03-16:
Hi @alexcorvis84, we are not yet planning any updates for RPC remote shell. But our team working on updating Gateway Remote Configuration widgets.

alexcorvis84, 2022-03-16:
> Hi @alexcorvis84, we are not yet planning any updates for RPC remote shell. But our team working on updating Gateway Remote Configuration widgets.

Ok, thanks for the quick feedback @samson0v !

alexcorvis84, 2022-04-04:
Hi @samson0v 

I'm wondering where can I find the source code of Gateway Widgets in official [ThingsBoard Github](https://github.com/thingsboard/thingsboard). I'd like to review the code about how they implement (rpc? rest api? mqtt?) the configuration updates.

Thanks in advance.

samson0v, 2022-04-08:
Hi @alexcorvis84, you can find them in the next paths:
[image]

eydonmlau, 2022-04-12:
@samson0v Hi, Is any possible to add 'remoteShell' config in Gateway widget?

samson0v, 2022-04-13:
Hi @eydonmlau, no. But our team working on updating Gateway Remote Configuration widgets.

yohnniebabe, 2022-05-25:
Hi @samson0v , I've tried using the gateway widget to configure the Tb gateway, but once I turn remote configuration on, the gateway wont work, saying config_folder_path is missing, is there any documentation on where I should add this info in the tb_gateway.yaml file?

samson0v, 2022-05-30:
Hi @yohnniebabe, it seems that you are using old Gateway version 2.9, please update it to the newest release and try again.

yohnniebabe, 2022-05-30:
> Hi @yohnniebabe, it seems that you are using old Gateway version 2.9, please update it to the newest release and try again.

Thanks! I updated to the latest gateway version and its working now!

closed 2022-08-08 by imbeacon

alexcorvis84, 2022-08-16:
Hi @imbeacon @samson0v 

I've been doing some testing over sending configuration through the widget.

As far as I've checked and related to this [Issue](https://github.com/thingsboard/thingsboard/issues/6380), once User click the **Save** button on Widget,  3 POST requests are done.

Then the following happens: 2 Shared attributes are created (**configuration** y **RemoteLogginLevel**) and 1 Server-Side attribute (configuration_drafts with value "e30=") is created.

As far as I understand, when IoT Gateway is restarted next time, it should load the configuration saved as **configuration** parameter (which contains the **new** configuration saved by the Widget in BASE64 string).

I've found that i.e when I change the **Maximum event packet size** to 5000, the configuration contains it as **"read_records_count":5000** parameter. As far as I understand this new value should be changed once IoT Gateway is restarted BUT but it is having no effect at all. 

I mean and If I check my **tb_gateway.yaml** file, I can see it has been modified (according to date and hour) but **"read_records_count"** has not change to new value.

[image]

Could you please double-check on your side? it might be a BUG?

I've not been able to modify **host** parameter as well through Gateway Widget :-(

yohnnylim, 2022-10-03:
Hi @samson0v ,

Is there a way we can configure custom converter using the gateway widget?

samson0v, 2022-10-05:
@yohnniebabe, no for now.