thingsboard/thingsboard-gateway #777: [HELP] Response to device when use telemetry http api

debacs83, 2022-03-16:
**Describe the issue**
I want to send a small string response ['81'] in reply to the device when it sends a JSON telemetry data with HTTP to Thingsboard. Device will check if the response from Thingsboard is the desired one or not. If it is so (here: '81') it will 
send the next message. Else it will try to send the same message again and again.  

I came to know that in PE there is some option (Integration Feature). But currently I am using CE. Now from Thingsboard github forum I found that thingsboard iot gateway may solve this issue if the message is small. 
(https://thingsboard.io/docs/iot-gateway/config/rest/)

But not sure how to do it. Any help would be very much appreciated.

With guidance from here, I could install TB IoT gateway. 

```
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-03-16 10:12:01 JST; 7s ago
   Main PID: 595610 (python3)
      Tasks: 1 (limit: 9463)
     Memory: 11.5M
     CGroup: /system.slice/thingsboard-gateway.service
             mq595610 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

 3月 16 10:12:01 spf systemd[1]: Started ThingsBoard Gateway.
```

**Configuration** (Attach your configuration file)
Sending my (1) tb_gateway.yaml, (2) rest.json, (3) mqtt.json in attachment.

**Connector name (If you need help with some connector/converter):**
ThingsBoard IoT Gateway REST connector.

**Versions (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - Thingsboard version [3.3.2]
 - Thingsboard IoT Gateway version [3.0.1-1]
 - Python version[3.8
[iot_gw.zip](https://github.com/thingsboard/thingsboard-gateway/files/8258369/iot_gw.zip)

samson0v, 2022-03-17:
Hi @debacs83, we will add this functionality in the nearest feature.
Thanks for your interest in ThingsBoard IoT Gateway!

debacs83, 2022-03-18:
Thank you for your response. That would be great.

closed 2022-03-18 by imbeacon