thingsboard/thingsboard-gateway #779: [BUG] Getting error when writting boolean/coils values to PLC through RPC / Modbus

mcavalcante, 2022-03-17:
**Describe the bug**
I´ve created a "Switc control" widget to set a boolean variable in my PLC, but I´m getting the error "IndexError: list index out of range" when I set the switch to off. When I set it to on, I don´t see any error, the variable is wrongly set to 0 on the PLC.

My modbus.json is set like this in the section "rpc":
```
        "rpc": [
...
         {
           "tag": "setValue",
           "type": "bits",
           "functionCode": 5,
           "objectsCount": 1,
           "address": 40017
         },
```

**Connector name (If bug in the some connector):**
modbus

**Error traceback (If available):**
```
Mar 17 00:04:38 thingsboard-gw python3[117983]: ""2022-03-17 00:04:38" - |ERROR| - [modbus_connector.py] - modbus_connector - server_side_rpc_handler - 477 - list index out of range"
Mar 17 00:04:38 thingsboard-gw python3[117983]: Traceback (most recent call last):
Mar 17 00:04:38 thingsboard-gw python3[117983]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 464, in server_side_rpc_handler
Mar 17 00:04:38 thingsboard-gw python3[117983]:     self.__process_rpc_request(server_rpc_request, rpc_command_config)
Mar 17 00:04:38 thingsboard-gw python3[117983]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 486, in __process_rpc_request
Mar 17 00:04:38 thingsboard-gw python3[117983]:     converted_data = device.config[DOWNLINK_PREFIX + CONVERTER_PARAMETER].convert(rpc_command_config,
Mar 17 00:04:38 thingsboard-gw python3[117983]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py", line 106, in convert
Mar 17 00:04:38 thingsboard-gw python3[117983]:     builder = builder[0]
Mar 17 00:04:38 thingsboard-gw python3[117983]: IndexError: list index out of range
```

**Versions (please complete the following information):**
OS: Ubuntu Desktop 20.04.3 and Ubuntu Server 20.04.4 for the ThingBoard Gateway
Thingsboard IoT Gateway version 3.0.1
Python version 3.8.10

**Troubleshooting / notes:**
I don´t know if it is the correct approach, but that one below worked for me. After changing the method 'convert' in the file /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py, inserting the 2nd and the 3rd lines below, the method switch control started to work properly:

```
            if variable_size / 8 > 1.0:
                if (not isinstance(value, str)):
                    value = [value] * 8
                builder_functions["bits"](bytes(value, encoding='UTF-8')) if isinstance(value, str) else \
                    builder_functions["bits"](bytes(value))
```

Scenario:
[image]

Switch Control:
[image]

samson0v, 2022-03-17:
Hi @mcavalcante, try to set `"type": coil` in your config file.
Thanks for your interest in ThingsBoard IoT Gateway!

mcavalcante, 2022-03-17:
> Hi @mcavalcante, try to set `"type": coil` in your config file. Thanks for your interest in ThingsBoard IoT Gateway!
Hello @samson0v ,

Tried, no luck. See:
[image]

[image]

having a deeper look at the file bytes_modbus_downlink_converter.py, maybe it didn´t work because the type coil is handled in the same "if" of the code.
[image]

samson0v, 2022-03-21:
@mcavalcante thanks, we will test and let you know about the result.

mcavalcante, 2022-03-23:
> @mcavalcante thanks, we will test and let you know about the result.

Thank you Samson0v. If you need some wireshark captures for the "good and bad" scenarios, pls let me know.

mcavalcante, 2022-04-02:
> **roubleshooting / notes:** I don´t know if it is the correct approach, but that one below worked for me. After changing the method 'convert' in the file [...]

Hello Andhuooo, have you tried it? Does it work for you?

"I don´t know if it is the correct approach, but that one below worked for me. After changing the method 'convert' in the file /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py, inserting the 2nd and the 3rd lines below, the method switch control started to work properly:

            if variable_size / 8 > 1.0:
                if (not isinstance(value, str)):
                    value = [value] * 8
                builder_functions["bits"](bytes(value, encoding='UTF-8')) if isinstance(value, str) else \
                    builder_functions["bits"](bytes(value))"

Andhuooo, 2022-04-07:
Hi  @mcavalcante ,  i don't understood the way you tried  .  i'm using docker  to run tb- gateway where  did i get this file "bytes_modbus_downlink_converter.p" for editing

samson0v, 2022-07-18:
Hi @mcavalcante, any updates? Can we close the issue?

closed 2022-10-13 by imbeacon

memind-rad, 2023-03-17:
Hi @samson0v,

I´m still getting this error with the version below:
```
""2023-03-17 10:35:13" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2023-03-17 10:35:13" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2023-03-17 10:35:14" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - v3.2"
""2023-03-17 10:35:14" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -

[===UPDATE===]

 New version v3.2 is available!
```

[===UPDATE===]
"

With the original code, I can red the bool variables in the PLC, but I can´t write bool in the same variables. 

In this case I also needed to patch with the code suggested here in this thread (below), and worked:
```
        if variable_size / 8 > 1.0:
            if (not isinstance(value, str)):
                value = [value] * 8
            builder_functions["bits"](bytes(value, encoding='UTF-8')) if isinstance(value, str) else \
                builder_functions["bits"](bytes(value))"
```
 BR