thingsboard/thingsboard-gateway #781: [HELP] IoT Gateway configuration - Advanced security: TLS + Private Key

alexcorvis84, 2022-03-17:
**Describe the issue**

I'm trying to test connection of the Thingsboard IoT Gateway to Thingsboard platform via **Advanced security: TLS + Private Key** **without** success :-( 

There's no log message that could help me debug what's happening. Please check my configuration and 'error' traceback log below attached.

I wonder if the port should be instead 8883? There's no info about it in the online documentation for this case, so I'm not sure if it must be changed or not.

_Please note FYI:_ The connection of the gateway through Access Token is working ok but I need to setup TLS + Private Key.

**Configuration**
```
thingsboard:
  maxPayloadSizeBytes: 8192
  host: test-device2.hostname.example
  port: 1883
  remoteShell: true
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    privateKey: /home/user/key.pem
    caCert: /home/user/tb-server-chain.pem
    cert: /home/user/cert.pem
#     accessToken: ****
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_serial.json
```

**Error traceback (If it was raised):**
```
""2022-03-17 08:50:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
""2022-03-17 08:50:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
```

**Versions (please complete the following information):**
 - OS: **Raspbian GNU/Linux 10 (buster)**
 - Thingsboard IoT Gateway version **3.0.1**
 - Python version **3.7.3**

samson0v, 2022-03-17:
Hi @alexcorvis84, please, set the log level to DEBUG instead of INFO in all places in logs.conf file and send these logs.
Thanks for your interest in ThingsBoard IoT Gateway!

alexcorvis84, 2022-03-17:
> 

Hello @samson0v below my l**ogs.conf** file modified to DEBUG instead of INFO:

```
[loggers]
keys=root, service, connector, converter, tb_connection, storage, extension
[handlers]
keys=consoleHandler, serviceHandler, connectorHandler, converterHandler, tb_connectionHandler, storageHandler, extensionHandler
[formatters]
keys=LogFormatter
[logger_root]
level=ERROR
handlers=consoleHandler
[logger_connector]
level=DEBUG
handlers=connectorHandler
formatter=LogFormatter
qualname=connector
[logger_storage]
level=DEBUG
handlers=storageHandler
formatter=LogFormatter
qualname=storage
[logger_database]
level=DEBUG
handlers=databaseHandler
formatter=LogFormatter
qualname=database
[logger_tb_connection]
level=DEBUG
handlers=tb_connectionHandler
formatter=LogFormatter
qualname=tb_connection
[logger_service]
level=DEBUG
handlers=serviceHandler
formatter=LogFormatter
qualname=service
[logger_converter]
level=DEBUG
handlers=converterHandler
formatter=LogFormatter
qualname=converter
[logger_extension]
level=DEBUG
handlers=connectorHandler
formatter=LogFormatter
qualname=extension
[handler_consoleHandler]
class=StreamHandler
level=DEBUG
formatter=LogFormatter
args=(sys.stdout,)
[handler_connectorHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/connector.log", "d", 1, 7,)
[handler_storageHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/storage.log", "d", 1, 7,)
[handler_databaseHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/database.log", "d", 1, 7,)
[handler_serviceHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/service.log", "d", 1, 7,)
[handler_converterHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/converter.log", "d", 1, 3,)
[handler_extensionHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/extension.log", "d", 1, 3,)
[handler_tb_connectionHandler]
level=DEBUG
class=logging.handlers.TimedRotatingFileHandler
formatter=LogFormatter
args=("/var/log/thingsboard-gateway/tb_connection.log", "d", 1, 3,)
[formatter_LogFormatter]
format="%(asctime)s - |%(levelname)s| - [%(filename)s] - %(module)s - %(funcName)s - %(lineno)d - %(message)s" 
datefmt="%Y-%m-%d %H:%M:%S"
```

After stopping the thingsboard-gateway.service and start it again, the service log (/var/log/thingsboard-gateway/service.log) gives the same information as previously :(

[image]

[image]

alexcorvis84, 2022-03-17:
@samson0v

**FYI:**

I tried to 'connect'  and publish a message through **mosquitto_pub** with success, but I needed to add **--insecure** flag

_Example:_ 
`
mosquitto_pub --insecure --cafile tb-server-chain.pem -d -q 1 -h "test-device2.hostname.example" -p "8883" -t "v1/devices/me/telemetry" --key key.pem --cert cert.pem -m {"temperature":100}`

So I wonder if there's some way to  setup this '--insecure' mode on IoT Gateway configuration file?

I am commenting this because at least this way it has been able to connect and send data, but I am not sure if it is related to the problem of not being able to connect to the platform.

Hope it helps for your support and feedback.

samson0v, 2022-03-17:
@alexcorvis84, you have to configure ThingsBoard following [this guide](https://thingsboard.io/docs/user-guide/mqtt-over-ssl/).

alexcorvis84, 2022-03-17:
> @alexcorvis84, you have to configure ThingsBoard following [this guide](https://thingsboard.io/docs/user-guide/mqtt-over-ssl/).

Hi @samson0v. 

Thanks. Yes, I have my ThingsBoard instance configured as described on guide. 

Please check below my config, I'm running the ThingsBoard through **docker run**:

`sudo docker run -it -p 9090:9090 -p 1883:1883 -p 7070:7070 -p 5683-5688:5683-5688/udp -p 8883:8883 -v ~/.mytb-data:/data -v ~/.mytb-logs:/var/log/thingsboard -v ~/certs:/config --name mytb --restart always --env-file environment_vars.txt thingsboard/tb-cassandra`

**Environmental variables** are being passed on a text file (environment_vars.txt), which content is as follows:
```
MQTT_SSL_ENABLED=true
MQTT_SSL_CREDENTIALS_TYPE=PEM
MQTT_SSL_PEM_CERT=/config/server.pem
MQTT_SSL_PEM_KEY=/config/server_key.pem
MQTT_SSL_BIND_PORT=8883
MQTT_SSL_SKIP_VALIDITY_CHECK_FOR_CLIENT_CERT=true
```

I've been making tests to try to somehow debug what's happening. My last discovery is that if I submit the **mosquitto_pub** command described previously BUT **without** --insecure flag, this are the errors showed on Thingsboard logs:

```
2022-03-17 12:14:56,324 [nioEventLoopGroup-4-5] ERROR o.t.s.t.mqtt.MqttTransportHandler - [5a9d07d6-114b-459e-a3e4-c0b98a1aff10] Unexpected Exception
io.netty.handler.codec.DecoderException: javax.net.ssl.SSLHandshakeException: Insufficient buffer remaining for AEAD cipher fragment (2). Needs to be more than tag size (16)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:477)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:276)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: javax.net.ssl.SSLHandshakeException: Insufficient buffer remaining for AEAD cipher fragment (2). Needs to be more than tag size (16)
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:349)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:292)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:287)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:134)
	at java.base/sun.security.ssl.SSLEngineImpl.decode(SSLEngineImpl.java:681)
	at java.base/sun.security.ssl.SSLEngineImpl.readRecord(SSLEngineImpl.java:636)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:454)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:433)
	at java.base/javax.net.ssl.SSLEngine.unwrap(SSLEngine.java:637)
	at io.netty.handler.ssl.SslHandler$SslEngineType$3.unwrap(SslHandler.java:295)
	at io.netty.handler.ssl.SslHandler.unwrap(SslHandler.java:1341)
	at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1234)
	at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1283)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:507)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:446)
	... 17 common frames omitted
Caused by: javax.crypto.BadPaddingException: Insufficient buffer remaining for AEAD cipher fragment (2). Needs to be more than tag size (16)
	at java.base/sun.security.ssl.SSLCipher$T13GcmReadCipherGenerator$GcmReadCipher.decrypt(SSLCipher.java:1868)
	at java.base/sun.security.ssl.SSLEngineInputRecord.decodeInputRecord(SSLEngineInputRecord.java:240)
	at java.base/sun.security.ssl.SSLEngineInputRecord.decode(SSLEngineInputRecord.java:197)
	at java.base/sun.security.ssl.SSLEngineInputRecord.decode(SSLEngineInputRecord.java:160)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:111)
	... 28 common frames omitted
```

Any feedback is welcome. Many thanks in advance

alexcorvis84, 2022-03-25:
Hello @samson0v 

Can you please provide any feedback with new logs? Thanks in advance

samson0v, 2022-07-18:
Hi @alexcorvis84, is this issue still actual for you?

alexcorvis84, 2022-07-26:
> Hi @alexcorvis84, is this issue still actual for you?

Hi @samson0v

Yes, it is! I'm still have not succeed to test connection through TLS + Private Key :(

Any feedback regarding the new logs I attached previously? Thanks in advance.

samson0v, 2022-07-26:
@alexcorvis84, don't know what the problem is, I test TLS + Private Key connection a week ago and all worked fine for me with Gateway 3.1 and ThingsBoard Cloud.
Use this [instruction](https://thingsboard.io/docs/user-guide/certificates/) to generate cert and key

alexcorvis84, 2022-07-26:
> @alexcorvis84, don't know what the problem is, I test TLS + Private Key connection a week ago and all worked fine for me with Gateway 3.1 and ThingsBoard Cloud. Use this [instruction](https://thingsboard.io/docs/user-guide/certificates/) to generate cert and key

Ok, will check **_again_** and give you an update once done. Thanks

alexcorvis84, 2022-08-03:
Hi @samson0v 

I've been able today to make some test again and I can confirm it's working now.

Not sure, but I updated my TB platform to latest 3.4 release and remade again all the steps...and it worked!

Just a comment that would help other users making tests:

I needed to add the flag _--insecure_ to mosquitto_pub in order to disable verification of the server hostname in the server certificate ([info](https://mosquitto.org/man/mosquitto_sub-1.html))

`mosquitto_pub --cafile tb-server-chain.pem -d -q 1 -h "YOUR_TB_HOST" -p "8883" \
-t "v1/devices/me/telemetry" --key key.pem --cert cert.pem -m {"temperature":25} --insecure`

[image]
[image]

samson0v, 2022-08-03:
Hi @alexcorvis84, glad to hear everything is working! Can I close the issue?

alexcorvis84, 2022-08-03:
> Hi @alexcorvis84, glad to hear everything is working! Can I close the issue?

Yeah, sure! Thanks for your support.

closed 2022-08-03 by samson0v