thingsboard/thingsboard-gateway #785: not generating log files  [HELP]

akashagre, 2022-03-24:
[image]

Thingsboard gateway not generating log file

samson0v, 2022-03-28:
Hi @akashagre, it will be unable to save and see the logs because of the STREAM ONLY mode of your virtual machine.
Please, use the docker image if you are using Windows.

imbeacon, 2022-05-30:
Hi @akashagre ,

The possible reasons for this issue are:
1. logs folder didn't created, if you installed the gatesway as a service please check folder /var/log/thingsboard-gateway - is it exists? If you installed the gateway from the sources - please create logs folder in the thingsboard-gateway folder.
2. Some issue with permissions - the gateway in case of working like a service uses user thingsboard_gateway to write logs - please check are permissions for writing logs allowed to user thingsboard_user for logs folder.

closed 2022-05-30 by imbeacon