thingsboard/thingsboard-gateway #786: Unable to connect through opc ua also not generating errors.[HELP]

akashagre, 2022-03-24:
[image]
[image]

samson0v, 2022-03-28:
Hi @akashagre, it will be unable to see the logs because of the STREAM ONLY mode of your virtual machine.
Please, use the docker image if you are using Windows.

Andhuooo, 2022-03-30:
Hi , can you  show the code of opcua server or  can you please connect it with thw prosys opcua client and share the  window screen shot .

samson0v, 2022-04-08:
@akashagre, I can't help you because of your `STREAM ONLY` mode of your virtual machine, it is makes reading logs unavailable.

samson0v, 2022-05-23:
@akashagre, a situation like this may appear because the gateway cannot write logs to the log file. The most common reasons are:
- logs folder doesn't exist (Using the version from the repository you should create the logs folder manually in the root of the repository folder, in another case please check if the path to the logs folder exists)
- thingsboard_gateway user doesn't have permission to write to the logs folder.

samson0v, 2022-07-18:
Hi @akashagre, any updates? Can we close the issue?

samson0v, 2022-08-04:
Hi @akashagre, any updates?

closed 2022-08-16 by imbeacon