thingsboard/thingsboard-gateway #787: [HELP] How to call an exterbak REST API in localhost of my ThingsBoard gateway

mcavalcante, 2022-03-25:
**Describe the issue**
In my ThingsBoard gateway I have a API responding in the localhost that aims to establish a SSH session to an external address (public address). This API is called through the URL http://127.0.0.1:5000/todo/api/v1.0/inicia, where the 'inicia' is the the "word" that triggers some methods that internally forms a SSH session. For this API to work, I send the parameters in JSON format, through POST request, like the one below:
{
   "usuario_ssh": "useer",
   "servidor_ssh": "server",
   "senha_ssh": "password",
   "porta_ssh": "11000",
   "tunnels": [
       {
         "server_port": 11001,
         "destination_address": "10.0.0.102:502",
         "descricao": "PLC-300-1 IP"
        },
       {
         "server_port": 44002,
         "destination_address": "127.0.0.1:3240",
         "descricao": "PLC-300-1 USB"
        },
       {
         "server_port": 44003,
         "destination_address": "127.0.0.1:80",
         "descricao": "RaspBerry Web Server"
        }
   ]
}

I trying to find out how to call this API from ThingBoard platform, but I couldn´t find an way so far. Reading the document https://thingsboard.io/docs/user-guide/rpc/ and https://thingsboard.io/docs/iot-gateway/getting-started/, it seems that the connector request.json would be the apropriate way for this task, by the means of the the RPC section. but it isn´t clear to me how to call this API. 

In my understanding, through the document, is that I should call the API in my gateway via ThingBoard server, using something like the link below:

Format:
curl -v -X POST -d @set-gpio-request.json http://localhost:8080/api/plugins/rpc/twoway/$DEVICE_ID \
--header "Content-Type:application/json" \
--header "X-Authorization: $JWT_TOKEN"

Command that I´m using:
curl --insecure -v -X POST -d "{'usuario_ssh':'user','servidor_ssh':'serverone','senha_ssh':'passsabc','porta_ssh':'11000','tunnels':[{'server_port':11001,'destination_address':'10.0.0.102:502','descricao':'PLC-300-1 IP'},{'server_port':11002,'destination_address':'127.0.0.1:3240','descricao':'PLC-300-1 USB'},{'server_port':11003,'destination_address':'127.0.0.1:80','descricao':'RaspBerry Web Server'}]}" http://thingsboard:8080/api/plugins/rpc/twoway/3b159050-aa4f-11ec-9622-77b107538394 --header "Content-Type:application/json" --header "X-Authorization: Bearer eOWZmMTQ2MTFjZSIsImZpcnN0TmFtZSI6Ik1HIEdlbmVyYWwgQXV0b21hdGlvbiIsImxhc3ROYW1lIjoiIiwiZW5hYmxlZCI6dHJ1ZSwiaXNQdWJsaWMiOmZhbHNlLCJ0ZW5hbnRJZCI6Ijk0M2ZiNzQwLTliMmQtMTFlYy04ZmExLTZkOWZmMTQ2MTFjZSIsImN1c3RvbWVySWQiOiIxMzgxNDAwMC0xZGQyLTExYjItODA4MC04MDgwODA4MDgwODAiLCJpc3MiOiJ0aGluZ3Nib2FyZC5pbyIsImlhdCI6MTY0ODE2OTI2NywiZXhwIjoxNjQ4MTc4MjY3fQ.xsEzlknH6fnRhHqqRrfZEXGE9S6vZfU4eMMb1iC_tWsBuwu1PM2dcA6qq8BnTOAK-_0g5hnbzpgKPOv2KO1gOA"

but this request is returning me the following output:
{"status":400,"message":"Invalid request body","errorCode":31,"timestamp":"2022-03-25T03:04:55.154+00:00"}* Closing connection 0

I don´t know what I´m wrong. Could you please give me some clue?

Notice that the API http://127.0.0.1:5000/todo/api/v1.0/inicia works properly if I call it from some REST client.

samson0v, 2022-03-28:
Hi @mcavalcante, you can use [RPC requests](https://thingsboard.io/docs/iot-gateway/config/rest/#server-side-rpc-section). For debugging you can use a special dashboard widget in ThingsBoard as below
[image]

samson0v, 2022-04-04:
Hi @mcavalcante, any updates, can we close the issue?

mcavalcante, 2022-04-04:
It is working now!

closed 2022-04-04 by mcavalcante