thingsboard/thingsboard-gateway #789: [HELP] Unable to build Docker image

l33tkrew, 2022-03-25:
**Describe the issue**
I use latest files on this github repository. When I want to build image through "docker build -t myimage ." command and when I want to run a container with that image, I got some errors. 

Note: I copy the Dockerfile into main directory. Also, I did not  modify any file. I directly cloned this repository into my computer.

Here is the log files when I want to start the container (based on the image I built above)
**Error traceback (If it was raised):**
```
Traceback (most recent call last):
  File "//./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 27, in <module>
    from simplejson import JSONDecodeError, dumps, load, loads
ModuleNotFoundError: No module named 'simplejson'
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - Thingsboard IoT Gateway version [latest github version]
 - Python version[e.g. 3.9]

samson0v, 2022-03-28:
Hi @l33tkrew, we recommend using the [official docker image ](https://thingsboard.io/docs/iot-gateway/install/docker-installation/)anyway.
In your case, try to add `simplejson` to the `requirements.txt` file.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-04-04:
Hi @l33tkrew, any updates, can we close the issue?

closed 2022-04-06 by l33tkrew