thingsboard/thingsboard-gateway #790: [HELP] 'NoneType' issue appears with current gateway version, but not in version 2.8

katjesandcoding, 2022-03-25:
I use the thingsboard IoT gateway in a Docker Desktop container with several odbc connectors.

They work (mostly) fine when I use the code for version 2.8 and the data is transmitted to the thingsboard server. However, when I use later version, the following error messages appear and the data can not be processed. Can anyone help, what might be the reason for this diffrence?

**Connector name :**
ODBC Connector

**Error traceback:**
""2022-03-25 14:32:35" - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 258 - [prod_cont ODBC Connector] Failed to process database row: 'NoneType' object has no attribute 'timestamp'"

""2022-03-25 14:32:35" - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 258 - [prod_cont ODBC Connector] Failed to process database row: 'NoneType' object has no attribute 'timestamp'"

""2022-03-25 14:32:35" - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 258 - [prod_cont ODBC Connector] Failed to process database row: 'NoneType' object has no attribute 'timestamp'"

""2022-03-25 14:32:35" - |INFO| - [odbc_connector.py] - odbc_connector - __poll - 229 - [prod_cont ODBC Connector] Polling iteration finished. Processed rows: current 3, total 13797"

**Versions :**
 - OS: Docker Desktop on Windows
 - Thingsboard IoT Gateway version  3.0.1 and 2.8
 - Python version 3.9.9

samson0v, 2022-03-28:
Hi @katjesandcoding, send, please, your `odbc.json` config file, also send the logs with `DEBUG` level (replace all `INFO` to `DEBUG` in `logs.conf` file).
Thanks for your interest in ThingsBoard IoT Gateway!

cwinkelmann, 2022-03-30:
Hi this behaviour can be reproduced using this code: https://gitlab.com/christian.winkelmann/thingsboard-data-transfer
I think this behaviour relates to https://github.com/thingsboard/thingsboard-gateway/issues/465

Error above happens when there is no "ts" column available. For instance if the odbc_connector config looks like this

```json
  "polling": {
    "query": "SELECT temperature, humidity, NOT_ts_BUT_SOMEOTHER_COLUMN FROM db.CUBE_DATA_1 WHERE ts > ? ORDER BY ts ASC LIMIT 200",
    "period": 30,
    "iterator": {
      "column": "ts",
      "query": "SELECT MIN(ts) - 1 FROM db.CUBE_DATA_1",
      "persistent": false
    }

  }
```

If this is corrected other issues occur. The connector logs are in the repo:
```
""2022-03-30 11:06:36" - |INFO| - [odbc_connector.py] - odbc_connector - __poll - 229 - [cw cubes 1 ODBC Connector] Polling iteration finished. Processed rows: current 2, total 122"
""2022-03-30 11:07:06" - |INFO| - [odbc_connector.py] - odbc_connector - __poll - 229 - [cw cubes 2 ODBC Connector] Polling iteration finished. Processed rows: current 2, total 124"
""2022-03-30 11:07:06" - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 258 - [cw cubes 1 ODBC Connector] Failed to process database row: 'int' object has no attribute 'timestamp'"
""2022-03-30 11:07:06" - |WARNING| - [odbc_connector.py] - odbc_connector - __process_row - 258 - [cw cubes 1 ODBC Connector] Failed to process database row: 'int' object has no attribute 'timestamp'"
```
"cw cubes 1 ODBC Connector" fails instantly because it gets the data directly from a BIGINT column.
"cw cubes 2 ODBC Connector" seems to work but passes the data to thingsboard wrong:

[image]

If you run the code given the above setup it shows.

So questions are:
- What is the necessary column type: "TIMESTAMP" would be best, it seems, BIGINT and DATETIME should work to though
- Is there something missing/wrong with odbc config?