thingsboard/thingsboard-gateway #795: [HELP] Access IP outside the Docker container

emrekardaslar, 2022-04-04:
**Describe the issue**
I created a docker image and set the options to make configurations outside the container following your instructions. I want to access a port that is accessible by the host machine. How can I do that? For example: In tb_gateway.yaml, I want to enter an IP which is accessible from the host machine, how can I do that? I know it is more of a Docker question but if you can help me it would be great! Thank you.

**Versions (please complete the following information):**
 - OS: Wİndows
 - Thingsboard IoT Gateway version: 3.0.1
 - Python version: 3.9.5

samson0v, 2022-04-08:
Hi @emrekardaslar, please, send your Docker file and/or command you run it.
Thanks for your interest in ThingsBoard IoT Gateway!

emrekardaslar, 2022-04-08:
Actually, the docker file is the same as the existing one (https://github.com/thingsboard/thingsboard-gateway/blob/master/docker/Dockerfile)

I tried to run several commands:
* docker run --network host -it -v C:/Users/TEST/tb-gateway/config:/thingsboard_gateway/config -v C:/Users/TEST/tb-gateway/extensions:/thingsboard_gateway/extensions -v C:/Users/TEST/tb-gateway/logs:/thingsboard_gateway/logs --name ndu-gateway-2 --restart always gateway
* docker run --add-host="localhost:IP_TO_ACCESS" -it -v C:/Users/TEST/tb-gateway/config:/thingsboard_gateway/config -v C:/Users/TEST/tb-gateway/extensions:/thingsboard_gateway/extensions -v C:/Users/TEST/tb-gateway/logs:/thingsboard_gateway/logs --name ndu-gateway-2 --restart always gateway

I used MQTT config as:
{
  "broker": {
    "port": 1883,
"host": "127.0.0.1",
    "maxMessageNumberPerWorker": 20,
    "maxNumberOfWorkers": 200
  },
  "mapping": [ .....}

I know that I can reach 127.0.0.1 using host.docker.internal, I want to use different IPs.

Thanks for replying!

samson0v, 2022-04-13:
Hi @emrekardaslar, if I understand you in the right way, you have to use the next [command](https://docs.docker.com/engine/reference/builder/#expose)

closed 2022-05-20 by imbeacon