thingsboard/thingsboard-gateway #796: [HELP] Modbus Error at __function_to_device

emrekardaslar, 2022-04-05:
**Issue**

I'm trying to use the Modbus connector. To use it, I downloaded a Modbus slave emulator from https://www.modbustools.com/modbus_slave.html . I also downloaded the master emulator and checked that its working. After that, I slightly modified modbus.json and used it. I am getting an error: |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 438 - Exception Response(132, 4, IllegalFunction)" I believe that something is not right about my configuration so I wanted to ask you.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name:**
Modbus connector

**Error traceback (If it was raised):**
```
|ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 438 - Exception Response(132, 4, IllegalFunction)"
|ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(132, 4, IllegalFunction)"
```

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.9
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/8416322/tb_gateway.txt)
[modbus.txt](https://github.com/thingsboard/thingsboard-gateway/files/8416323/modbus.txt)

emrekardaslar, 2022-04-05:
The error is caused by the emulation.

jiang-sss, 2022-05-22:
> The error is caused by the emulation.

I also have this problem, could you please help to explain it

hylitjob, 2022-09-26:
I also have this problem, could you please help to fix it!

`""2022-09-26 07:11:31" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(132, 4, IllegalFunction)"
`

hylitjob, 2022-09-26:
```
{
  "master": {
    "slaves": [
      {
        "host": "100.100.140.75",
        "port": 5021,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "Temp Sensor",
        "sendDataOnlyOnChange": true,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "aaa",
            "type": "8int",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 1
          }
        ]
      }
    ]
  }
}

```

hylitjob, 2022-09-26:
> "slaves": [
>       {
>         "host": "100.100.140.75",
>         "port": 5021,
>         "type": "tcp",
>         "method": "socket",

How to solve this problem?