thingsboard/thingsboard-gateway #798: [BUG] Sending RPC over MQTT to device has singleQuotes in JSON

andrevocat, 2022-04-08:
Iot Plattform sends valid json wth doubleQuotes 

Log from gateway:
[Thread-4        ] [connector       ] [server_side_rpc_handler] - Publishing to: aaaa/sensor/ssssss/request/config/4 with data {'telemetryInterval': 1}

Messing on extenal mqtt broker is also: {'telemetryInterval': 1} <- should be doubleQuotes for valid JSON

We drilled down the Problem to tb-gateway in thingsboard-gateway/thingsboard_gateway/connectors/mqtt/mqtt_connector.py → Line 686: (_client ist PAHO-client and not TBDeviceMqttClient)

self.__log.info("Publishing to: %s with data %s", request_topic, data_to_send)
// HERE missing convertion of data_to_send back to JSON  -> data_to_send = json.dump(data_to_send)
self._client.publish(request_topic, data_to_send, retain=rpc_config.get('retain', False))

**Versions (please complete the following information):**
 tb-gateway 3.0.1

andrevocat, 2022-04-08:
Maybe there are othwer places missing this conversation, we didnt checked that fully.

samson0v, 2022-04-08:
Hi @andrevocat, we will test it and let you know about the results.
Thanks for your interest in ThingsBoard IoT Gateway!

closed 2022-04-08 by imbeacon