thingsboard/thingsboard-gateway #799: [HELP] Staggering the Start of Polling Interval for Modbus

cowherdboy, 2022-04-08:
**Describe the issue**
I have a Modbus Master that is connected to multiple slaves and set to poll every 1 minute. The modbus connector file is attached. When I do a poll, I noticed that many of the devices are intermittent, e.g. some of them respond 5 mins later, some take hours to get a reading. However, when I change the connector file to only 1 device then I consistently get data every minute. When multiple devices are polled I frequently get the following error:

2022-04-08 16:26:20,439 - ERROR - [modbus_connector.py] - modbus_connector - 288 - Connection lost! Reconnecting...

My suspicion is that the Modbus Master is unable to poll so many devices simultaneously. Is there a way to add an offset to the pollperiod for each device? In other words, I would like to poll device 1 at time 0, device 2 at time 5s and then repeat device 1 at 60s and device 2 at 65s.

**Configuration** (Attach your configuration file)
[modbus_v2.zip](https://github.com/thingsboard/thingsboard-gateway/files/8450152/modbus_v2.zip)

**Connector name (If you need help with some connector/converter):**
Modbus Connector

**Error traceback (If it was raised):**
N/A

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.8.10

samson0v, 2022-08-29:
Hi @cowherdboy, we made some modbus connector optimization, so try to use the Gateway version from the master branch and let us know about the result.

samson0v, 2022-10-01:
Hi @cowherdboy, any updates? Can we close the issue?