thingsboard/thingsboard-gateway #810: [HELP] No module named 'thingsboard_gateway'

santobedi, 2022-04-19:
**Describe the issue**
Unable to install Thingsboard IoT gateway in Raspberry Pi 3 (install from Source).  The last few lines of the log file during installation are attached. 

[image]

While running` python3 ./thingsboard_gateway/tb_gateway.py`, I get the following,

[image]

Note that the IoT gateway (installed from source) was running fine in the same device earlier. I had the reflash the Raspberry Pi due to some reason and attempted to install the IoT gateway. Is it a BUG?

santobedi, 2022-04-19:
@imbeacon  @samson0v,

samson0v, 2022-04-19:
Hi @santobedi, to install Gateway from the source you need to follow the next steps:
1. git clone https://github.com/thingsboard/thingsboard-gateway.git
2. cd thingsboard-gateway
3. python3 -m venv venv
4. source venv/bin/activate
5. python3 setup.py install
6. mkdir logs
7. python3 ./thingsboard_gateway/tb_gateway.py

Also, you may need to run the next command:
`pip install --upgrade protobuf`

Thanks for your interest in ThingsBoard IoT Gateway!

santobedi, 2022-04-19:
Hello @samson0v ,

Thanks for the prompt response. I followed the mentioned steps. However, still I get `ModuleNotFoundError: No module named 'thingsboard_gateway'`. Nothing seems changed with the new procedure.

samson0v, 2022-04-19:
@santobedi send, pls, the logs

santobedi, 2022-04-19:
@samson0v 
```

(venv) iots@iot2057:~/thingsboard-gateway $ sudo python3 setup.py install
[sudo] password for iots:
running install
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/setuptools/command/easy_install.py:160: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing thingsboard_gateway.egg-info/PKG-INFO
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/thingsboard_gateway
creating build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/odbc.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/rest.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/modbus.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/tb_gateway.yaml -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/mqtt.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/custom_serial.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/bacnet.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/request.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/ble.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/opcua.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/socket.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/grpc_connector_1.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/can.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/modbus_serial.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/ftp.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/logs.conf -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
copying build/lib/thingsboard_gateway/config/snmp.json -> build/bdist.linux-armv7l/egg/thingsboard_gateway/config
creating build/bdist.linux-armv7l/egg/thingsboard_gateway/tb_client
copying build/lib/thingsboard_gateway/tb_client/tb_gateway_mqtt.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/tb_client
copying build/lib/thingsboard_gateway/tb_client/__init__.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/tb_client
copying build/lib/thingsboard_gateway/tb_client/tb_device_mqtt.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/tb_client
creating build/bdist.linux-armv7l/egg/thingsboard_gateway/storage
creating build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/file_event_storage.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/event_storage_files.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/file_event_storage_settings.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/event_storage_reader_pointer.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/__init__.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/event_storage_reader.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
copying build/lib/thingsboard_gateway/storage/file/event_storage_writer.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/file
creating build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/memory
copying build/lib/thingsboard_gateway/storage/memory/memory_event_storage.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/memory
copying build/lib/thingsboard_gateway/storage/memory/__init__.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage/memory
copying build/lib/thingsboard_gateway/storage/event_storage.py -> build/bdist.linux-armv7l/egg/thingsboard_gateway/storage
[... cut ...]
(venv) iots@iot2057:~/thingsboard-gateway $ mkdir logs
mkdir: cannot create directory ‘logs’: File exists
(venv) iots@iot2057:~/thingsboard-gateway $ python3 ./thingsboard_gateway/tb_gateway.py
Traceback (most recent call last):
  File "/home/iots/thingsboard-gateway/./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
ModuleNotFoundError: No module named 'thingsboard_gateway'
(venv) iots@iot2057:~/thingsboard-gateway $

```

samson0v, 2022-04-19:
@santobedi hmmm, it seems that I was wrong, sorry.
You have to install Gateway without using venv. So deactivate it and delete venv/ directory.

1. git clone https://github.com/thingsboard/thingsboard-gateway.git
2. cd thingsboard-gateway
3. sudo python3 setup.py install
4. mkdir logs
5. python3 ./thingsboard_gateway/tb_gateway.py

santobedi, 2022-04-20:
Hello @samson0v,

I attempted installing gateway from `pip`. However, I get the following error,

### ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33'  not found (required by /usr/local/lib/python3.9/dist-packages/grpc/_cython/cygrpc.cpython-39-arm-linux-gnueabihf.so)

Moreover, the error regarding `ModuleNotFoundError: No module named 'thingsboard_gateway'` still persists when I try installing it from the source.

santobedi, 2022-04-20:
**UPDATE**

I solved the GLIBC 2.33 issue in my edge computer and the installed gateway (DEB) is working.

samson0v, 2022-04-20:
Hi @santobedi, it is a very strange behaviour of Gateway installation from the source because I tested the steps I described above and everything goes fine on my notebook under Linux Ubuntu.
I think it is some specific of your OS.
Will we resolve to continue it or deb installation resolves your problem?

santobedi, 2022-04-20:
Hello @samson0v ,

For now, deb installation solves my problem. I will also try installing from the source, if the error persists, I will update it here.

closed 2022-04-20 by santobedi