thingsboard/thingsboard-gateway #812: Modbus to MQTT converter using TB gateway!

santobedi, 2022-04-19:
Hello @imbeacon and @samson0v,

I need to use the data from smart sensors (that follow Modbus protocol) for the ML routine inside the edge computer where the Thingsboard IoT gateway is installed. I've written a python script to convert Modbus to MQTT, where the ML routine subscribes to MQTT topics (the Modbus polling results), does some computations, and publishes the results under the next topic. I've configured the `mqtt.json ` file inside Thingsboard-gateway based on the published results topic so that I can see the results as the telemetry in the Thingsboard server. 

As the TB IoT gateway has a Modbus connector that 'polls' data from devices and sends it to the TB server using MQTT, I think it is possible to subscribe to the topics that the Modbus connector is 'publishing'. Could you please suggest any ways of using TB IoT gateway as modbus2mqtt?

Thank you.

samson0v, 2022-04-20:
Hi @santobedi, if I understood you correctly, you can try to use this [API](https://thingsboard.io/docs/pe/reference/gateway-mqtt-api/#introduction).

santobedi, 2022-04-20:
Hello @samson0v ,

Thank you for the info. I will give it a try.

closed 2022-04-20 by santobedi