thingsboard/thingsboard-gateway #814: Downstreaming changes to the Python SDK?

jelacious, 2022-04-25:
Would it be possible to downstream the changes made in the repository to the [shared files](https://github.com/thingsboard/thingsboard-gateway/tree/master/thingsboard_gateway/tb_client) of the [Python SDK](https://github.com/thingsboard/thingsboard-python-client-sdk/)? Alternatively, is it possible to include those updated files as part of a "third party integration" in some neat way? I'd like to have the library be up-to-date and seeing as the Python SDK isn't actively being maintained right now I am unsure what to do.

From a maintenance point of view it seems a bit strange to include those files in this repository as well instead of specifying them as either a git submodule or including them via something like pip. I think a git submodule would be the neatest solution as then you can propagate updates very fast between the two repos, but updating the pip versions is also probably a good idea.

samson0v, 2022-04-26:
Hi @jelacious, we will update these methods in Python SDK in the nearest future.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-04-27:
Hi @jelacious, we updated these methods yesterday, so you can clone the newest version from the master branch.

jelacious, 2022-04-27:
Nice! 

Will there be a bump in version on [pip](https://pypi.org/project/tb-mqtt-client/) as well?

samson0v, 2022-04-27:
@jelacious yes, but I don't know when exactly this will be done.

jelacious, 2022-04-27:
Alright, thanks for the info! I'd suggest making the Python SDK repository a git submodule of this repository so that it will be easier to get updates between the two repositories.

Thanks for the help! <3
Слава Украине!

samson0v, 2022-04-27:
@jelacious we will discuss this suggestion, thanks.
Героям слава!

samson0v, 2022-05-24:
Hi @jelacious, we decided to use Python SDK as a base for Gateway, so all products will be in updated status.

closed 2022-09-07 by imbeacon