thingsboard/thingsboard-gateway #819: [BUG] SNMP connector does not send all telemetry

farmerkz, 2022-05-04:
**Describe the bug**
My SNMP connector config:

```
{
  "devices": [
    {
      "deviceName": "ArzRouter",
      "deviceType": "snmp",
      "ip": "192.168.1.254",
      "port": 161,
      "pollPeriod": 5000,
      "community": "public",
      "attributes": [
        {
          "key": "sysDescr",
          "method": "get",
          "oid": "1.3.6.1.2.1.1.1.0",
          "timeout": 6
        }
      ],
      "telemetry": [
        {
          "key": "eth1-beeline",
          "method": "get",
          "oid": "1.3.6.1.2.1.2.2.1.8.1",
          "timeout": 1
        },
        {
          "key": "l2tp-beeline",
          "method": "get",
          "oid": "1.3.6.1.2.1.2.2.1.8.6",
          "timeout": 1
        }
      ]
    }
  ]
}

```
On thingsboard I only get the second telemetry value.

**Connector name (If bug in the some connector):**
SNMP Connector

**Versions (please complete the following information):**
 - OS: 20.04.4 LTS
 - Thingsboard IoT Gateway version v.3.3.4
 - Python version 3.8.10

aedahh, 2022-06-04:
Hello @farmerkz, sorry for posting to this ticket. From your post, looks like getting snmp telemetry is working for your, so hope you can share some insights. I am new to Thingsboard and also try to get snmp from telemetry from local on premise device. I am running an iot-gateway locally from docker image and try to send telemetry data to my https://thingsboard.cloud instance. I don't know if this configuration is supported as I always get mqtt connection error and snmp time out. 
[image]

Are you running a local Thingsboard instance or cloud?
Can you share your configuration files and uplink and downlink converter code to parse snmp telemetry? 
Looking forward to your reply. Thanks

farmerkz, 2022-06-12:
Hi @aedahh !
I am using a local Thingsboard server and the tb-gateway is also on the same network as the server.

snmp.json:

```
{
  "devices": [
    {
      "deviceName": "ArzRouter",
      "deviceType": "snmp",
      "ip": "192.168.1.254",
      "port": 161,
      "pollPeriod": 5000,
      "community": "public",
      "attributes": [
        {
          "key": "sysDescr",
          "method": "get",
          "oid": "1.3.6.1.2.1.1.1.0",
          "timeout": 6
        }
      ],
      "telemetry": [
        {
          "key": "eth1-beeline",
          "method": "get",
          "oid": "1.3.6.1.2.1.2.2.1.8.1",
          "timeout": 6
        }
      ]
    },
    {
      "deviceName": "ArzRouter",
      "deviceType": "snmp",
      "ip": "192.168.1.254",
      "port": 161,
      "pollPeriod": 5000,
      "community": "public",
      "attributes": [
        {
          "key": "sysDescr",
          "method": "get",
          "oid": "1.3.6.1.2.1.1.1.0",
          "timeout": 6
        }
      ],
      "telemetry": [
        {
          "key": "l2tp-beeline",
          "method": "get",
          "oid": "1.3.6.1.2.1.2.2.1.8.6",
          "timeout": 1
        }
      ]
    }
  ]
}
```

ts4iot, 2023-07-12:
I think this issue/bug is not really solved, it is only fix by a configuration workaround. The snmp.json from @farmerkz first post:

```
"telemetry": [
  {
    "key": "eth1-beeline",
    "method": "get",
    "oid": "1.3.6.1.2.1.2.2.1.8.1",
    "timeout": 1
  },
  {
    "key": "l2tp-beeline",
    "method": "get",
    "oid": "1.3.6.1.2.1.2.2.1.8.6",
    "timeout": 1
  }
]
```

should be possible because the element "telemetry" is an array, and it makes no sense to create multiple device entries to transfer multiple oids from one device. The move of the following code snipped (see:  file snmp_connector.py) out of the loop leads to the problem that only the last element of the array will be transmitted to the platform:

```
if isinstance(converted_data, dict) and (converted_data.get("attributes") or converted_data.get("telemetry")):
    self.collect_statistic_and_send(self.get_name(), converted_data)
```

After moving the code snipped again in to the loop all values will be transmitted but then I got the following error:

|ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 861 - 'str' object has no attribute 'get'"

The error occures in the following section of function validate_converted_data(data) in file tb_utility.py:

```
if data.get("telemetry") is not None:
    for entry in data.get("telemetry"):
        if (entry.get("ts") is not None and len(entry.get("values")) > 0) or entry.get("ts") is None:
            got_telemetry = True
            break
```
This should be fixed.

Yassinebouothman, 2024-01-27:
Hey @farmerkz, do you know how can I use SNMP in the cloud version of thingsboard not the one installed locally? Thanks.