thingsboard/thingsboard-gateway #825: [HELP] How to connect multiple slave devices under the modbus-serial (rtu) connector

plm55555, 2022-05-11:
**Describe the issue**
Create description about your issue, and your actions to solve it.
I created a gateway device and two subdevices on the things board platform and set their association as: the gateway device contains two subdevices.Their attribute values, telemetry values, and shared attribute values are also configured in the profile modbus_new.json.When I configure any other device in the modbus_new.json that always works fine, but when I configure both devices simultaneously, I get the error.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
{
  "master": {
    "slaves": [
      {
        "unitId": 1,
        "deviceName": "testUpdate",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "temperature",
            "type": "bytes",
            "functionCode": 3,
            "registerCount": 1,
            "address": 256
          }
        ],
        "timeseries": [
          {
            "tag": "distance",
            "type": "bytes",
            "functionCode": 3,
            "objectsCount": 123,
            "address": 256
          }
        ],
	"attributeUpdates": [
          {
            "tag": "shared_value_1",
            "type": "32uint",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 512
          },
	  {
            "tag": "shared_value_2",
            "type": "32uint",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 513
          }
        ],
        "rpc": [
          {
            "tag": "bearing_bpfo",
            "type": "32uint",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 516
          }
        ],
        "host": null,
        "port": "/dev/ttyUSB0",
        "type": "serial",
        "method": "rtu",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": null,
        "retryOnEmpty": null,
        "retryOnInvalid": null,
        "baudrate": 9600,
        "pollPeriod": 5000,
	"connectAttemptCount":1
      },
      {
        "unitId":2,
        "deviceName":"RpcTest",
        "attributesPollPeriod":5000,
        "timeseriesPollPeriod":5000,
        "sendDataOnlyOnChange":false,
        "attributes":[
           {
            "byteOrder":"BIG",
            "tag":"temperature",
            "type":"bytes",
            "functionCode":3,
            "registerCount":1,
            "address":256
             }
           ],
        "timeseries":[
           {
            "tag":"distance",
            "type":"bytes",
            "functionCode":3,
            "objectsCount":123,
            "address":256
            }
          ],
        "attributeUpdates":[
            {
            "tag":"shared_value_1",
            "type":"32uint",
            "functionCode":6,
            "objectsCount":2,
            "address":512
            }
           ],
        "host":null,
        "port":"/dev/ttyUSB0",
        "type":"serial",
        "method":"rtu",
        "timeout":35,
        "byteOrder":"BIG",
        "wordOrder":"BIG",
        "retries":null,
        "retryOnEmpty":null,
        "retryOnInvalid":null,
        "baudrate":9600,
        "pollPeriod":5000,
        "connectAttemptCount":5
       }
    ]
  },
  
 "slave": null
}

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
Modbus Connector

**Error traceback (If it was raised):**
```
""2022-05-11 15:28:10" - |DEBUG| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 87 - datatype: telemetry 	 key: distance 	 value: None"
""2022-05-11 15:28:10" - |DEBUG| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 92 - {'deviceName': 'testUpdate', 'deviceType': 'default', 'telemetry': [], 'attributes': []}"
""2022-05-11 15:28:10" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 83 - Modbus Error: [Input/Output] device reports readiness to read but returned no data (device disconnected or multiple access on port?)"
NoneType: None
""2022-05-11 15:28:10" - |DEBUG| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 87 - datatype: telemetry 	 key: distance 	 value: None"
""2022-05-11 15:28:10" - |DEBUG| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 92 - {'deviceName': 'RpcTest', 'deviceType': 'default', 'telemetry': [], 'attributes': []}"
""2022-05-11 15:28:10" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 83 - Modbus Error: [Input/Output] device reports readiness to read but returned no data (device disconnected or multiple access on port?)"
NoneType: None
```

**Versions (please complete the following information):**
 - OS: [ Raspberry Pi ]
 - Thingsboard IoT Gateway version [3.0.1]
 - Python version[3.9.2]

plm55555, 2022-05-11:
@imbeacon @samson0v  Can you take up some time to help me solve this problem？
Thanks a lot !

TheSamabo, 2022-05-11:
Hi
`device disconnected or multiple access on port?` Could  mean that multiple processes are trying to read serial port at the same time.  Do you have any other process/program that is attached to that exact serial port?

Another thing that could cause this is that you are setting `"host": none`  on each slave. Try deleting that line its not necessary. 
you can checkout docs https://thingsboard.io/docs/iot-gateway/config/modbus/?modbusConnection=serial for setting up serial  modbus communication
Sam

plm55555, 2022-05-20:
I have already solved this problem.

closed 2022-05-20 by plm55555