thingsboard/thingsboard-gateway #827: [BUG] version `GLIBC_2.29' not found

dh-jy, 2022-05-13:
**Describe the bug**
i installed thingsboard-gateway 3.0.1.post0 . when i run the gateway, I have the problem in the picture below

[image]

**Versions (please complete the following information):**
 - OS: [e.g. raspberrypi]
 - Thingsboard IoT Gateway version [e.g. 3.0.1.post0]
 - Python version[e.g. 3.7]

samson0v, 2022-05-13:
Hi @dh-jy, you have to install GRPCIO
`sudo pip3 install grpcio`
Thanks for your interest in ThingsBoard IoT Gateway!

dh-jy, 2022-05-13:
Hi，i have installed it, but the error is still
 [image]

dh-jy, 2022-05-13:
It seems that the latest version has this problem。Version 2.9 doesn't seem to have this problem

samson0v, 2022-05-30:
Hi @dh-jy, any updates? Can we close the issue?

samson0v, 2022-08-03:
Hi @dh-jy, can we close the issue? or issue is still actual for you?

closed 2022-08-04 by dh-jy