thingsboard/thingsboard-gateway #829: [HELP] Gateway not reconnecting to Modbus device after disconnection

jessedcpc, 2022-05-19:
**Describe the issue**
Upon loss of connection with my Modbus device, TB Gateway requires a restart to start communicating again

**Configuration**
[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/8723262/modbus.json.txt)
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/8723263/tb_gateway.yaml.txt)

**Connector name:**
Modbus Connector

**Versions:**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway v2.9
 - Thingsboard v3.2

samson0v, 2022-06-30:
Hi @jessedcpc,
you have to add the next parameters to your device configuration:
```
"connectAttemptTimeMs": 5000,
"connectAttemptCount": 5,
"waitAfterFailedAttemptsMs": 300000,
"retries": true,
"retryOnEmpty": true,
"retryOnInvalid": true,
```

samson0v, 2022-06-30:
Also, it is recommended to update your Gateway to the newest version (3.1) via the master branch or other suitable for you way.
Modbus Connector was remastered and fixed a lot of bugs in the new version.

samson0v, 2022-08-04:
Hi @jessedcpc, can we close the issue?

closed 2022-08-16 by imbeacon