thingsboard/thingsboard-gateway #834: [Feature] Beckhoff ADS connector

JavierNR, 2022-05-25:
Hello!

I'm trying to develop a new connector for thingsboard-gateway here: 
[https://github.com/JavierNR/thingsboard-gateway/tree/feature/ads-connector](https://github.com/JavierNR/thingsboard-gateway/tree/feature/ads-connector)

This will support connecting to TwinCAT plc's from Beckhoff using pyads python library, at the moment my connector doesn't work as I need to do some changes (not a python expert here). Maybe there is a possibility for tb-gateway devs help here ;). I can test with an end device if needed. 
Let's discuss it here, also I can do a PR with my work at the moment....

JavierNR, 2022-06-06:
Many thanks @samson0v , at the moment my developed library connects successfully to an end PLC and maybe it's possible for me to help tb-devs to read variables to an end device.
I'm a little bit lost when dealing with tb-gateway converters and how to implement them regarding pyads, so if devs provide me with some kind of template/scafold maybe I can extend it with my limited knowledge...
Also I can put real end device under my IP if you want.

Best regards!

samson0v, 2022-06-10:
Hi @JavierNR, for providing a converter you have to implement the next code blocks:
1. Implement a function that will load default converters and custom converters, for example in many connector it see like this:
[image]
2. Save loaded converter in the config, for example:
[image]
3. Simply call the `convert` method when you want to convert data:
[image]

And for a converter example, you can use the next (MQTT Converter):
1. thingsboard_gateway/connectors/mqtt/mqtt_uplink_converter.py
2. thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py

and the next example (FTP Converter):
1. thingsboard_gateway/connectors/ftp/ftp_converter.py
2. thingsboard_gateway/connectors/ftp/ftp_uplink_converter.py

or any other you want (we have a lot))

samson0v, 2022-08-05:
Hi @JavierNR, any updates? Maybe you need some help with connector implementation?

samson0v, 2026-01-14:
@JavierNR 📢 a little announcement:
We just launched our [official Discord server](https://discord.com/invite/mJxDjAM3PF), we'd love to see you there!