thingsboard/thingsboard-gateway #835: [HELP]

wangpony, 2022-05-26:
**Describe the issue**
Create description about your issue, and your actions to solve it.
Simulate the Modbus TCP protocol data. The 32int and 32float data received by the gateway are inconsistent with the sent data

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2022-05-30:
Hi @wangpony, please, send your config for the modbus connector and full log.
Also, describe the steps to reproduce your bug.
Thanks for your interest in ThingsBoard IoT Gateway!

wangpony, 2022-06-01:
My issue has been solved，thank you