thingsboard/thingsboard-gateway #836: [HELP]How to cancel remote-logging

plm55555, 2022-05-27:
**Describe the issue**
I did this feature https://thingsboard.io/docs/iot-gateway/guides/how-to-enable-remote-logging/ in my project but now I find this operation makes all logs saved to the database with too high memory and I want to cancel this feature , so how can I do it.
Thank you for your reply.

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Versions (please complete the following information):**
- OS: [ Raspberry Pi ]
- Thingsboard IoT Gateway version [3.0.1]
- Python version[3.9.2]
[image]

imbeacon, 2022-05-27:
Hi @plm55555 ,

The simplest way to disable remote logging is to remove "RemoteLoggingLevel" attribute from the gateway.
Or change it to higher level? like ERROR, the you will receive only error messages.

plm55555, 2022-05-30:
>Thanks for your reply.  When I delete “RemoteLoggingLevel” this property, although the logging will not be saved to the database again ,but I can't check it in Raspberry Pi again. Before I add this attribute, I can view the DEBUG-level attributes in the Raspberry Pi in the logs file, which is exactly what I need. So how should I do it？
[image]
@imbeacon

imbeacon, 2022-06-30:
Hi @Lanruo555 ,

Actually after removing this parameter you can restart the gateway and configuration will be taken from logs.conf file in your config folder, as before adding this parameter.

samson0v, 2022-07-18:
Hi @Lanruo555, any updates? Can we close the issue?

closed 2022-08-16 by imbeacon