thingsboard/thingsboard-gateway #839: [BUG] after update from 3.0.1 to 3.1 gateway did not start on ubuntu  /socketserver.py AttributeError: module 'socket' has no attribute 'AF_INET'

hansju-11, 2022-05-29:
I did an update from 3.0.1 to 3.1 with  ...latest/download/python3-thingsboard-gateway.deb 
now my gateway canot start any more.

sysadmin@thingsboard:/var/lib/thingsboard_gateway$ /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 16, in <module>
    import logging.config
  File "/usr/lib/python3.8/logging/config.py", line 37, in <module>
    from socketserver import ThreadingTCPServer, StreamRequestHandler
  File "/usr/lib/python3.8/socketserver.py", line 390, in <module>
    class TCPServer(BaseServer):
  File "/usr/lib/python3.8/socketserver.py", line 437, in TCPServer
    address_family = socket.AF_INET
AttributeError: module 'socket' has no attribute 'AF_INET'
sysadmin@thingsboard:/var/lib/thingsboard_gateway$

sysadmin@thingsboard:/var/lib/thingsboard_gateway$ python3 --version
Python 3.8.10
DISTRIB_DESCRIPTION="Ubuntu 20.04.4 LTS"

hansju-11, 2022-05-30:
found solution !!
1. backup postgres sql and conf  (thingsboard and thingsboard-gateway) folder
2. install fresh ubuntu
3. install thingsboard 3.3.4.1
4. restore sql db and thingsboard conf folder
5. install thingsboard-gateway from deb.
6. got an error:
May 30 08:22:00 thingsboard systemd[1]: Started ThingsBoard Gateway.
May 30 08:22:00 thingsboard python3[12391]: Traceback (most recent call last):
May 30 08:22:00 thingsboard python3[12391]:   File "<string>", line 1, in <module>
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
May 30 08:22:00 thingsboard python3[12391]:     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line                                                   49, in <module>
May 30 08:22:00 thingsboard python3[12391]:     from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.                                                  py", line 18, in <module>
May 30 08:22:00 thingsboard python3[12391]:     from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager                                                  .py", line 24, in <module>
May 30 08:22:00 thingsboard python3[12391]:     from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkCo                                                  nverter
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_downlink_c                                                  onverter.py", line 21, in <module>
May 30 08:22:00 thingsboard python3[12391]:     from thingsboard_gateway.gateway.proto.messages_pb2 import *
May 30 08:22:00 thingsboard python3[12391]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/proto/messages_pb2.py", line                                                   22, in <module>
May 30 08:22:00 thingsboard python3[12391]:     create_key=_descriptor._internal_create_key,
May 30 08:22:00 thingsboard python3[12391]: AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'

7. sysadmin@thingsboard:~$ sudo pip install  protobuf==3.20.1
Collecting protobuf==3.20.1
  Downloading protobuf-3.20.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (1.0 MB)
     |████████████████████████████████| 1.0 MB 1.8 MB/s
Installing collected packages: protobuf
  Attempting uninstall: protobuf
    Found existing installation: protobuf 4.21.1
    Uninstalling protobuf-4.21.1:
      Successfully uninstalled protobuf-4.21.1
Successfully installed protobuf-3.20.1

8.) restor thingsboard-gateway conf folder.

---> ready

closed 2022-05-30 by hansju-11

wilterdinkrobert, 2022-10-04:
I'm getting exactly the same error as @ilseva reports when I build a clean deb package on the master branch. Doing the suggested rename solves the issue as well.

@samson0v