thingsboard/thingsboard-gateway #844: [HELP] Several BACnet devices on same IP and Port, need to use device identifier - BACnet router?

PaulHHolmes, 2022-06-12:
**Describe the issue**
Create description about your issue, and your actions to solve it.

I've configured quite a few bacnet devices in gateways now, but this is my first that is using a BACnet router (Siemens PXG3.L).
I have 4 devices which all are on IP 192.168.1.2 and port 47808

Here's what it looks like in a YABE scan, and YABE reads the data points perfectly.

[image]

I want to read data from device 2098177, but can't see any way of adding that into my device configuration - either in the device adress, or the objectId. Here's what I see when I hover in YABE on the device.

[image]

Does anyone know how to add the device identifier in please?

**Configuration** 
{
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "Device name here",
      **"deviceType": "default",
      "address": "192.168.1.2:47808",**
      "pollPeriod": 300000,
      "timeseries": [
        {
          "key": "BacnetTemp1",
          "objectId": "analogInput:34",
          "propertyId": "presentValue"
        }
      ]
    }
  ]
}

**Connector name**
Bacnet

**Error traceback (If it was raised):**

**Versions (please complete the following information):**
 - OS: Dietpi 8.5.1 (5.15.43-sunxi)
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.9

samson0v, 2022-08-02:
Hi @PaulHHolmes, it seems that the bacpypes library can't discover that variant of device connectivity.
Also, in my mind, this type of device connectivity (abstraction) does not make any practical sense.

Thanks for your interest in ThingsBoard IoT Gateway!

closed 2022-08-16 by imbeacon

knityanand, 2024-08-03:
@PaulHHolmes were you able to resolve this. Could you please help with the config if you were successful

HarvUK, 2024-08-04:
> @PaulHHolmes were you able to resolve this. Could you please help with the config if you were successful

Hi,
I couldn't solve this - the issue seems to be caused when devices are routed through other devices, so the main IP address is of the router, then each bacnet MS/TP, LONnet etc is represented on a different port.

Later version of YABE seem much better at handline this. It's a while since I've tried it on TBGW though.

Thanks,

Harv

knityanand, 2024-08-04:
OK. Got it.