thingsboard/thingsboard-gateway #845: [BUG] OPC UA connector doesn't install in docker version

AntoineLeLoarer, 2022-06-13:
**Describe the bug**

When I try to run the IoT gateway in a docker, I have a bug that say the Connector implementation not found ofr OPC-UA Connector.  

**Your Gateway Environment**

* own setup
  * My computer
  * ThingsBoard IoT Gateway Version : [3.0.1]
  * OS: Windows 10 Professionnel 19044.1526
  * Docker version : v20.10.12

**Your Device**

* Connectivity
  * Gateway OPC-UA
* Device vendor and model

**To Reproduce**
Steps to reproduce the behavior:
1. Run command : `docker run -it -v C:/Users/MyUSER/tb-gateway/config:/thingsboard_gateway/config -v C:/Users/MyUSER/tb-gateway/extensions:/thingsboard_gateway/extensions -v C:/Users/MyUSER/tb-gateway/logs:/thingsboard_gateway/logs --name tb-gateway --restart always thingsboard/tb-gateway`
2. Remove useless connector to get only tb_gateway.yaml, opcua.json, logs.conf, connected_devices.json
3. Edit tb_gateway.yaml to get : 
```thingsboard:
  host: 10.1.0.186
  port: 1813
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: FtWYNv49xXojszbhI9xv
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json
```
5. Edit opcua.json to get : 
```{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://localhost:4334/UA/serverTestAntoine",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.DeviceRandomGen",
        "deviceNamePattern": "AABBCCDDEEFF0011",
        "timeseries": [
          {
            "key": "temperature",
            "path": "${Root\\.Objects\\.DeviceRandomGen\\.RandomDoubleRO}"
          }
        ]
      }
    ]
  }
}
```

7. Re run the docker
8. See the erro in the logs

**Expected behavior**
Install OPC-UA connector and connet to my server

**LOGS DOCKER**

```
""2022-03-18 09:19:48" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
""2022-03-18 09:19:48" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
""2022-03-18 09:19:48" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
""2022-03-18 09:19:48" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
OPC-UA library not found
Collecting opcua
  Downloading opcua-0.98.13.tar.gz (572 kB)
     |████████████████████████████████| 572 kB 4.1 MB/s
  Preparing metadata (setup.py) ... done
Collecting lxml
  Downloading lxml-4.8.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (6.9 MB)
     |████████████████████████████████| 6.9 MB 12.3 MB/s
Collecting python-dateutil
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     |████████████████████████████████| 247 kB 10.8 MB/s
Collecting pytz
  Downloading pytz-2021.3-py2.py3-none-any.whl (503 kB)
     |████████████████████████████████| 503 kB 16.2 MB/s
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.9/site-packages/six-1.16.0-py3.9.egg (from python-dateutil->opcua) (1.16.0)
Building wheels for collected packages: opcua
  Building wheel for opcua (setup.py) ... done
  Created wheel for opcua: filename=opcua-0.98.13-py3-none-any.whl size=628151 sha256=9e45b5d74640cf53966a996071a56cf7a3e9fa2cebfd0af6549fac8c59123b63
  Stored in directory: /root/.cache/pip/wheels/f7/08/b8/3de089aa34284ac9970ba9118586aae9e887c07975a21da453
Successfully built opcua
Installing collected packages: pytz, python-dateutil, lxml, opcua
Successfully installed lxml-4.8.0 opcua-0.98.13 python-dateutil-2.8.2 pytz-2021.3
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 21.3.1; however, version 22.0.4 is available.
You should consider upgrading via the '/usr/local/bin/python -m pip install --upgrade pip' command.
""2022-03-18 09:19:53" - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - name 'Node' is not defined"
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/tb_utility/tb_loader.py", line 63, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 39, in <module>
    class OpcUaConnector(Thread, Connector):
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 509, in OpcUaConnector
    def get_node_path(self, node: Node):
NameError: name 'Node' is not defined
""2022-03-18 09:19:53" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 473 - Connector implementation not found for OPC-UA Connector"
""2022-03-18 09:19:53" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
```

samson0v, 2022-06-28:
Hi @AntoineLeLoarer, connector implementation is not fount because of an exception that has been raised (`NameError: name 'Node' is not defined`). 
It seems that opcua library install incorrectly. Try to restart your docker and container.

Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-06-30:
Hi @AntoineLeLoarer, any updates?

closed 2022-07-28 by imbeacon