thingsboard/thingsboard-gateway #852: [HELP] How to determine thingsboard-gateway version & run update service behind a proxy

cowherdboy, 2022-06-16:
Hello,

I would like to know how to do the following:

1) Determine the thingsboard-gateway version that I am running

2) I am behind a proxy and the update service is unable to connect to the update server. Do I need to specify the proxy address and port somewhere as I do with the thingsboard license server? 

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.1]
 - Python version 3.8

samson0v, 2022-06-28:
Hi @cowherdboy, 
1. you can see the Gateway version at the top of logs as in the example below:
    [image]
    or if you are using source code, you can see it in the `setup.py` file.
2. you can't specify the proxy for the update service (honestly, you do not need to do this, because apart from notifications there will be nothing more useful for you)

Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-06-30:
Hi @cowherdboy, can we close the issue?

cowherdboy, 2022-07-01:
Hi @samson0v, Thank you for your helpful feedback. The issue may be closed.

closed 2022-07-01 by cowherdboy