thingsboard/thingsboard-gateway #853: [BUG] Gateway Cannot Publish Data For All Devices

emrekardaslar, 2022-06-17:
**Describe the bug**
* Gateway cannot publish every data, some of the devices on the platform cannot recieve new data. Does changing storage type effect the data persistence? What type of storage should I use? I decided to use file storage since it is recommended for persistence. What are your recommendations?

**Connector name**
mqtt connector

**Versions (please complete the following information):**
 - OS: Ubuntu Server 2020.04
 - Thingsboard IoT Gateway version 2.0
 - Python version 3.7

samson0v, 2022-06-28:
Hi @emrekardaslar, can you, please, send your MQTT config file, data that send to Gateway and logs.

samson0v, 2022-06-30:
Hi @emrekardaslar, any updates?

samson0v, 2022-07-01:
Hi @emrekardaslar, try to change to True  `Overwrite activity time for connected device` for the Gateway device.
[image]

emrekardaslar, 2022-07-18:
* Unfortunately no. The OS may cause the issue but I'm not sure. If I can figure it out, I will update you.

samson0v, 2022-07-29:
Hi @emrekardaslar, please, update your Gateway to the newest version via the master branch. It seems that the #877 issue relates to yours and it was fixed.

closed 2022-07-30 by emrekardaslar