thingsboard/thingsboard-gateway #856: [HELP] Modbus values are not updated

otoase, 2022-06-21:
**Describe the issue**
I am trying to connect a ModBus device to ThingsBoard using the gateway. When starting the gateway, first value is successfully read and sent to the platform. After that, it is not updated anymore (nor even the stats).
As seen from traffic captures, Modbus value is read correctly. However, the value sent to ThingsBoard (MQTT message) is always the same.

**Configuration** (Attach your configuration file)

tb_gateway.yaml:
> thingsboard:
>   host: XXXXXX
>   port: 1883
>   remoteShell: false
>   remoteConfiguration: false
>   statsSendPeriodInSeconds: 3600
>   minPackSendDelayMS: 0
>   checkConnectorsConfigurationInSeconds: 60
>   security:
>     accessToken: XXXXXXXX
>   qos: 1
> storage:
>   type: memory
>   read_records_count: 10
>   max_records_count: 1000
> grpc: 
>   enabled: false
>   serverPort: 9595
>   keepaliveTimeMs: 10000  
>   keepaliveTimeoutMs: 5000
>   keepalivePermitWithoutCalls: true
>   maxPingsWithoutData: 0
>   minTimeBetweenPingsMs: 10000
>   minPingIntervalWithoutDataMs: 5000
> connectors:
>     name: Modbus Connector
>     type: modbus
>     configuration: modbus.json

modbus.json

> {
>   "master": {
>     "slaves": [
>       {
>         "host": "xxx.xxx.xxx.xxx",
>         "port": 502,
>         "type": "tcp",
>         "method": "socket",
>         "timeout": 35,
>         "byteOrder": "BIG",
> 	    "wordOrder": "BIG",
>         "retries": true, 
>         "retriesOnEmpty": true,
>         "retriesOnInvalid": true,
>         "pollPeriod": 10000,
>         "unitId": 1,
>         "deviceName": "MBTEST",
>         "connectAttemptTimeMs": 5000,
>         "connectAttemptCount": 5,
>         "waitAfterFailedAttemptsMs": 300000,
>         "sendDataOnlyOnChange": false,
>          "timeseries": [
>           {
>             "tag": "16int_read",
>             "type": "16int",
>             "functionCode": 3,
>             "objectsCount": 1,
>             "address": 1
>           }
>         ]
>       }
>     ]
>   }
> }

**Connector name (If you need help with some connector/converter):**
ModBus Connector

**Versions (please complete the following information):**
 - OS: Raspbian GNU/Linux 11
 - Thingsboard: v3.3.4.1
 - Thingsboard IoT Gateway: v3.1
 - Python: v3.9.2

dingbihuii, 2022-06-22:
Maybe you can try changing the values of modbus devices

samson0v, 2022-06-30:
Hi @otoase, update your Gateway to the newest version via the master branch and let us know if the problem still exists.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-07-01:
Hi @otoase, can you, please, send the logs (replace all INFO to DEBUG in logs.conf file in /config folder)?

ebarrieau, 2022-07-02:
I just installed Thingsboard and Thingsboard Gateway on a fresh Ubuntu 22.04 install on a physical server and I have this exact same issue. @otoase your work around to use sqlite as the storage type is working for me as well. If there are logs or configuration files that I can supply that would help identify the root of the issue I am happy to help.

samson0v, 2022-10-01:
Hi @otoase, there were some related bug fixes, so update your Gateway via the master branch and let us know if it helps.

closed 2022-11-01 by imbeacon