thingsboard/thingsboard-gateway #862: [BUG]connection FAIL with error 5 not authorised

WingedKitten, 2022-06-23:
**Describe the bug**
Shut down several times after the gateway appeared this problem.
Use PyCharm or daemon appeared this problem
Restart alone thingsboard services have not had any effect
When I restart thingsboard server returned to normal

**Connector name (If bug in the some connector):**
[mqtt Connector]

**Error traceback (If available):**
```
""2022-06-23 18:50:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 181 - Gateway started."
""2022-06-23 18:50:55" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 241 - MQTT Broker Connector connected to 192.168.0.198:1886 - successfully."
""2022-06-23 18:50:55" - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 146 - connection FAIL with error 5 not authorised"
```
thingsboard log find

`2022-06-16 19:27:08,469 [nioEventLoopGroup-4-9] ERROR o.t.s.t.mqtt.MqttTransportHandler - [20e0d52d-2bb7-4041-be93-95868445ac4e] Unexpected Exception
java.io.IOException: Connection reset
at java.base/sun.nio.ch.FileDispatcherImpl.read0(Native Method)
at java.base/sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39)
at java.base/sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:276)
at java.base/sun.nio.ch.IOUtil.read(IOUtil.java:233)
at java.base/sun.nio.ch.IOUtil.read(IOUtil.java:223)
at java.base/sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:356)
at io.netty.buffer.PooledByteBuf.setBytes(PooledByteBuf.java:253)
at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1132)
at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:350)
at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:151)
at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:719)
at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
at java.base/java.lang.Thread.run(Thread.java:829)`

**Versions (please complete the following information):**
 - OS: [centos7]
 - Thingsboard IoT Gateway version [3.0.1]
 - Thingsboard  version [3.3.2]
 - Python version[3.9]

samson0v, 2022-06-28:
Hi @WingedKitten, maybe you have several gateways connected to ThingsBoard, so it is the reason of this behaviour.

WingedKitten, 2022-06-28:
@samson0v Several gateway to connect to the ThingsBoard and used by the same token?
But before that, I run for two weeks of gateway also appeared the same problem, I can't solve,even if I change the token, in the end I restart all the servers.
Excuse me, how much thingsboard can support a gateway connection, not the same token,

samson0v, 2022-06-28:
@WingedKitten by default ThingsBoard allows only one Gateway but you can change the configuration of your ThingsBoard instance (parameter ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE)

WingedKitten, 2022-06-28:
At the same time, there is a magic, when I use the same server configuration files on my computer started Thingsboard service gateway connection is returned to normal, when I stopped the local service gateway began to prompt the error.
I don't understand why is this,In the end I restart all the servers

samson0v, 2022-06-28:
@WingedKitten, I think [this issue](https://github.com/thingsboard/thingsboard-gateway/issues/643) will be useful for you.

WingedKitten, 2022-06-28:
I think my description may be a little problem，I mean multiple gateway using different credentials

samson0v, 2022-06-28:
@WingedKitten by the way try to change parameter ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE

WingedKitten, 2022-06-28:
ok，I'll try.
And there are another issue, a gateway can withstand much equipment.Last week I made a simple stress tests.
Gateway can only support 10 device report the frequency of data per second.
So I found this part in your code, 'sleep (0.2)' is because Thingsboard frequency cannot afford too quick to report?
`sleep(0.2)
                            if success and self.tb_client.is_connected():
                                self._event_storage.event_pack_processing_done()
                                # del devices_data_in_event_pack
                                # devices_data_in_event_pack = {}`
If I have a lot of equipment need to use gateway,whether to need to use a lot of gateway to receive data.

samson0v, 2022-06-28:
@WingedKitten  sleep(.2) is an optimization so that python does not use 100% CPU

WingedKitten, 2022-06-28:
@samson0v Maybe I can reduce him to improve the efficiency of the gateway?

samson0v, 2022-06-28:
@WingedKitten you can, but it is not recommended (it is default construction when you use forever loop, you can google it).

samson0v, 2022-06-30:
Hi @WingedKitten, any updates?

WingedKitten, 2022-06-30:
Thank you for your answer, I am busy doing other things, this week have free time I'll test again

WingedKitten, 2022-07-10:
I changed the parameters of sleep, and now his processing speed faster, but the amount of CPU is also higher

closed 2022-07-10 by WingedKitten