thingsboard/thingsboard-gateway #864: [HELP] Connector implementation not found for OPC-UA Connector

vachannd, 2022-06-29:
**Describe the issue**
Create description about your issue, and your actions to solve it.

I have an instance of thingsboard-gateway running on UBUNTU 20.04 server with OPC-UA connector. The gateway cannot find the OPC-UA connector implementation. I have installed all the necessary libraries including and not limited to unixodbc, pyodbc, python asyncua and have msft odbc drivers installed as well.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

thingsboard:
  host: ENDPOINT_HERE
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: ACCESS_CODE_HERE
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
\#  type: file
\#  data_folder_path: ./data/
\#  max_file_count: 10
\#  max_read_records_count: 10
\#  max_records_per_file: 10000
\#  type: sqlite
\#  data_file_path: ./data/data.db
\#  messages_ttl_check_in_hours: 1
\#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
\#  -
\#    name: MQTT Broker Connector
\#    type: mqtt
\#    configuration: mqtt.json
\#
\#  -
\#    name: Modbus Connector
\#    type: modbus
\#    configuration: modbus.json
\#
\#  -
\#    name: Modbus Connector
\#    type: modbus
\#    configuration: modbus_serial.json
\#
\ -
   name: OPC-UA Connector
   type: opcua
   configuration: opcua.json
\#
\#  -
\#    name: BLE Connector
\#    type: ble
\#    configuration: ble.json
\#
\#  -
\#    name: REQUEST Connector
\#    type: request
\#    configuration: request.json
\#
\#  -
\#    name: CAN Connector
\#    type: can
\#    configuration: can.json
\#
\#  -
\#    name: BACnet Connector
\#    type: bacnet
\#    configuration: bacnet.json
\#
\#  -
\#    name: ODBC Connector
\#    type: odbc
\#    configuration: odbc.json
\#
\#  -
\#    name: REST Connector
\#    type: rest
\#    configuration: rest.json
\#
\#  -
\#    name: SNMP Connector
\#    type: snmp
\#    configuration: snmp.json
\#
\#  -
\#    name: FTP Connector
\#    type: ftp
\#    configuration: ftp.json

**Connector name (If you need help with some connector/converter):**

OPC-UA Connector

**Error traceback (If it was raised):**
```
'
""2022-06-29 21:48:43" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
""2022-06-29 21:48:43" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
""2022-06-29 21:48:43" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
""2022-06-29 21:48:43" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 473 - Connector implementation not found for OPC-UA Connector"
""2022-06-29 21:48:43" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
""2022-06-29 21:48:43" - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 146 - connection FAIL with error 5 not authorised"
'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version : 3.0.1 
 - Python version: 3.8

vachannd, 2022-06-30:
Going to close this issue but leave it on the repo to help out anyone who has this issue later. Turns out, it was an issue with importing one of the dependencies for OPC-UA. If the logs don't help with understanding where the problem is, the best way to solve any 'connector implementation not found error' could be opening up a python console and trying to import all the libraries needed for the specific connector.

closed 2022-06-30 by vachannd

samson0v, 2022-06-30:
Hi @vachannd, merged a new PR that log exception if connector implementation is not found. So clone the master branch and try it.
Thanks for your interest in ThingsBoard IoT Gateway!

DejonckheereWard, 2022-06-30:
I believe this is also related to this fix i posted a while ago.
https://github.com/thingsboard/thingsboard-gateway/issues/553