thingsboard/thingsboard-gateway #868: [BUG]

TTnTTT, 2022-07-05:
**Describe the bug**
 Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
[image]

**Connector name (If bug in the some connector):**
[e.g. Modbus Connector]

**config file
modbus.json:
```
{
  "master": {
    "slaves": [
      {
        "host": "192.168.100.69",
        "port": 8899,
        "type": "tcp",
        "method": "rtu",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 300000,
        "unitId": 6,
        "deviceName": "空调(DDSU666.004 型单相电子式电能表)",
        "deviceType": "ammeter_sensor",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 24,
        "waitAfterFailedAttemptsMs": 1800000,
        "attributes": [

        ],
        "timeseries": [
          {
            "tag": "A相电压（V）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8192
          },
          {
            "tag": "A相电流（A）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8194
          },
          {
            "tag": "瞬时总有功功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8196,
            "multiplier": 1000
          },
          {
            "tag": "瞬时总无功功率（var）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8198
          },
          {
            "tag": "瞬时总现有功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8200,
            "multiplier": 1000
          },
          {
            "tag": "总工功率因素",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8202
          },
          {
            "tag": "电网频率（Hz）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8206
          },
          {
            "tag": "有功总电能（kWh）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 16384
          }
        ],
        "attributeUpdates": [
        ],
        "rpc": [
        ]
      },
      {
        "host": "192.168.100.70",
        "port": 8899,
        "type": "tcp",
        "method": "rtu",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 300,
        "unitId": 8,
        "deviceName": "电脑(DDSU666.004 型单相电子式电能表)",
        "deviceType": "ammeter_sensor",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 24,
        "waitAfterFailedAttemptsMs": 1800000,
        "attributes": [

        ],
        "timeseries": [
          {
            "tag": "A相电压（V）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8192
          },
          {
            "tag": "A相电流（A）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8194
          },
          {
            "tag": "瞬时总有功功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8196,
            "multiplier": 1000
          },
          {
            "tag": "瞬时总无功功率（var）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8198
          },
          {
            "tag": "瞬时总现有功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8200,
            "multiplier": 1000
          },
          {
            "tag": "总工功率因素",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8202
          },
          {
            "tag": "电网频率（Hz）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8206
          },
          {
            "tag": "有功总电能（kWh）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 16384
          }
        ],
        "attributeUpdates": [
        ],
        "rpc": [
        ]
      },
      {
        "host": "192.168.100.71",
        "port": 8899,
        "type": "tcp",
        "method": "rtu",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 3000,
        "unitId": 9,
        "deviceName": "流量计(DDSU666.004 型单相电子式电能表)",
        "deviceType": "ammeter_sensor",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 24,
        "waitAfterFailedAttemptsMs": 1800000,
        "attributes": [

        ],
        "timeseries": [
          {
            "tag": "A相电压（V）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8192
          },
          {
            "tag": "A相电流（A）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8194
          },
          {
            "tag": "瞬时总有功功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8196,
            "multiplier": 1000
          },
          {
            "tag": "瞬时总无功功率（var）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8198
          },
          {
            "tag": "瞬时总现有功率（W）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8200,
            "multiplier": 1000
          },
          {
            "tag": "总工功率因素",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8202
          },
          {
            "tag": "电网频率（Hz）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 8206
          },
          {
            "tag": "有功总电能（kWh）",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 16384
          }
        ],
        "attributeUpdates": [
        ],
        "rpc": [
        ]
      }

    ]
  }
}

```

tb_gateway.yaml:
```
thingsboard:
  host: 127.0.0.1 # thingsboard.cloud
  port: 1883
  remoteShell: true
  remoteConfiguration: true
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 600
  security:
    accessToken: ZJRsFxjniCBK0KKA0pih # PUT_YOUR_GW_ACCESS_TOKEN_HERE ZJRsFxjniCBK0KKA0pih
  qos: 1
  maxPayloadSizeBytes: 102400
storage:
#  type: memory
#  read_records_count: 100
#  max_records_count: 100000
  type: file
  data_folder_path: ./data/
  max_file_count: 10
  max_read_records_count: 10
  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: true
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json
```

**Error traceback (If available):**
```
""2022-07-05 11:26:07" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response"
NoneType: None
```

**Versions (please complete the following information):**
 - OS: [Windows 11]
 - Thingsboard IoT Gateway version [3.1]
 - Python version[3.7.3]

zxz666, 2022-07-07:
hi
   I have the same issues,Pls tell me if you have solved it?
   ths.

samson0v, 2022-07-11:
Hi @TTnTTT, update your Gateway to the newest version via the master branch (there are fixes for the Modbus connector) and let us know if it helps.

Thanks for your interest in ThingsBoard IoT Gateway!

zxz666, 2022-07-11:
OK,THS！

samson0v, 2022-07-18:
Hi @TTnTTT and @zxz666, any updates?

TTnTTT, 2022-07-21:
> Hi @TTnTTT and @zxz666, any updates?

Sorry, haven't noticed the news recently. The gateway version I use is the latest tag3.1, do I need to use the source code of the master branch?

TTnTTT, 2022-07-21:
> Hi @TTnTTT and @zxz666, any updates?

I tried  the master branch code，but recive the same error.
error message like this:
[image]

TTnTTT, 2022-07-21:
> > Hi @TTnTTT and @zxz666, any updates?
> 
> I tried the master branch code，used same mudbus.conf,but recive the same error. Error information like this: [image] The last is error information.

I think because I used the same ip and port (192.168.100.71:8899) but different unitId (One is 5, the other is 9).

samson0v, 2022-07-22:
> > > Hi @TTnTTT and @zxz666, any updates?
> > 
> > 
> > I tried the master branch code，used same mudbus.conf,but recive the same error. Error information like this: [image] The last is error information.
> 
> I think because I used the same ip and port (192.168.100.71:8899) but different unitId (One is 5, the other is 9). Does Gateway handle this situation?

@TTnTTT Yes, Gateway handle this situation

samson0v, 2022-12-05:
Hi @TTnTTT, try to use the latest version of Gateway (v.3.2.), there were bug fixes (data process and support for pymodbus version 3.0) and let us know about the result.

samson0v, 2022-12-30:
@TTnTTT any updates?

closed 2023-01-26 by samson0v