thingsboard/thingsboard-gateway #871: [HELP] How to persist connected_devices.json

gregwinn, 2022-07-07:
**Describe the issue**
I noticed that when sending data through the gateway the name is collected from "deviceNameJsonExpression" as expected. However, when I provide a more user-friendly name in Thingsboard it will update the name on the gateway but if the Gateway is restarted or fails all data from connected_devices.json is lost. When sending data from the same device it simply creates a new device in Thingsboard. So now I have two of the same devices just with different names.

I would like to be able to persist connected_devices.json so that when users rename devices and the gateway is restarted all that data will not be lost. I am sure this has been solved so if someone can point me in the right direction that would be great!

**Configuration** (Attach your configuration file)
Using file storage right now:
```
storage:
  type: file
  data_folder_path: /data/thingsboard-gateway/
  max_file_count: 8
  max_read_records_count: 6
  max_records_per_file: 16
```

**Connector name (If you need help with some connector/converter):**
Using the MQTT Connector

**Versions (please complete the following information):**
 - OS: Running in Docker Ubuntu 20.04
 - Thingsboard IoT Gateway version: Latest (Installed Today July 6th, 2022)
 - Python version: 3.9

gregwinn, 2022-07-07:
Based on this [ThingsBoard IoT Gateway Features](https://thingsboard.io/docs/iot-gateway/how-device-removing-renaming-works/) - I should be getting an RPC on the Gateway to tell me of the rename.... Again that is all working however how can I intercept that method and send the name change off to the device?

I am trying to catch the method using this in my mqtt.json file
```
"serverSideRpc": [
        {
            "methodFilter": "gateway_device_renamed",
            "requestTopicExpression": "sensor/update/${deviceName}/request/${methodName}/${requestId}",
            "valueExpression": "${params}"
        }
]
```
But I am not getting any messages on that topic.

samson0v, 2022-07-11:
Hi @gregwinn, we have the parameter `handleDeviceRenaming` in the tb_gateway.yaml file, so set its value to `True`.
[image]

Thanks for your interest in ThingsBoard IoT Gateway!

gregwinn, 2022-07-11:
Thank you @samson0v

closed 2022-07-11 by gregwinn