thingsboard/thingsboard-gateway #872: I am working on ODBC connection. Mounted the tb gateway in docker and success. Created SQL server tables.[BUG]

shauser1, 2022-07-08:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

shauser1, 2022-07-08:
[logs1.txt](https://github.com/thingsboard/thingsboard-gateway/files/9071553/logs1.txt)
attached the logs

samson0v, 2022-07-11:
Hi @shauser1, it seems that you haven't installed the odbc driver, so install it following the next [instruction](https://github.com/mkleehammer/pyodbc/wiki/Install).
Also, this [issue](https://github.com/thingsboard/thingsboard-gateway/issues/863) can help to solve yours.

Thanks for your interest in ThingsBoad IoT Gateway!

shauser1, 2022-07-12:
I installed odbc driver and still receive the same error. Could you please suggest/help

b 'SQL Server' : file not found (0) (SQLDriverConnect)")"
""2022-07-11 05:22:14" - |INFO| - [odbc_connector.py] - odbc_connector - run - 172 - [ODBC Connector] Will reconnect to database in 60 second(s)"
""2022-07-11 05:23:14" - |DEBUG| - [odbc_connector.py] - odbc_connector - __init_connection - 294 - [ODBC Connector] Opening connection to database"
""2022-07-11 05:23:14" - |ERROR| - [odbc_connector.py] - odbc_connector - __init_connection - 322 - [ODBC Connector] Failed to connect to database: ('01000', "[01000] [unixODBC][Driver Manager]Can't open lib 'SQL Server' : file not found (0) (SQLDriverConnect)")"
""2022-07-11 05:23:14" - |INFO| - [odbc_connector.py] - odbc_connector - run - 172 - [ODBC Connector] Will reconnect to database in 60 second(s)"
""2022-07-11 05:24:14" - |DEBUG| - [odbc_connector.py] - odbc_connector - __init_connection - 294 - [ODBC Connector] Opening connection to database"
""2022-07-11 05:24:14" - |ERROR| - [odbc_connector.py] - odbc_connector - __init_connection - 322 - [ODBC Connector] Failed to connect to database: ('01000', "[01000] [unixODBC][Driver Manager]Can't open lib 'SQL Server' : file not found (0) (SQLDriverConnect)")"
""2022-07-11 05:24:14" - |INFO| - [odbc_connector.py] - odbc_connector - run - 172 - [ODBC Connector] Will reconnect to database in 60 second(s)"
""2022-07-11 05:25:14" - |DEBUG| - [odbc_connector.py] - odbc_connector - __init_connection - 294 - [ODBC Connector] Opening connection to database"
""2022-07-11 05:25:14" - |ERROR| - [odbc_connector.py] - odbc_connector - __init_connection - 322 - [ODBC Connector] Failed to connect to database: ('01000', "[01000] [unixODBC][Driver Manager]Can't open lib 'SQL Server' : file not found (0) (SQLDriverConnect)")"
""2022-07-11 05:25:14" - |INFO| - [odbc_connector.py] - odbc_connector - run - 172 - [ODBC Connector] Will reconnect to database in 60 second(s)"
""2022-07-11 05:26:14" - |DEBUG| - [odbc_connector.py] - odbc_connector - __init_connection - 294 - [ODBC Connector] Opening connection to database"
""2022-07-11 05:26:14" - |ERROR| - [odbc_connector.py] - odbc_connector - __init_connection - 322 - [ODBC Connector] Failed to connect to database: ('01000', "[01000] [unixODBC][Driver Manager]Can't open li

samson0v, 2022-07-12:
@shauser1, from your logs, I think, that you did not correctly config the path to the odbc driver. Maybe [this](https://stackoverflow.com/questions/38534154/linux-python3-cant-open-lib-sql-server) can help you. If not, try to google your error. From these logs, I can't exactly say what is wrong with your driver and database.

samson0v, 2022-07-14:
Hi @shauser1, any updates? Can we close the issue?

closed 2022-08-16 by imbeacon