thingsboard/thingsboard-gateway #877: [HELP] Limits on amount of data you can send through a single Gateway?

gregwinn, 2022-07-11:
**Describe the issue**
I have a very busy Gateway that gets about 80 to 150 messages a min. After looking at this issue [826](https://github.com/thingsboard/thingsboard-gateway/issues/826) I see they had to split up into multiple gateways. I am having the same issue with devices showing up in Thingsboard however telemetry data is not saved consistently. Just like in #826 I am getting data sometimes and other times it's just the device name and that's it.

I have checked the logs and everything looks normal (all logs are set to DEBUG) - I am not seeing any ERROR or WARNING.

I am using sqlite3 for storage on the gateway and that seems to be working well - no errors.

**Configuration** (Attach your configuration file)
```
thingsboard:
  host: mythingsboardUrl.com
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: TOKEN
storage:
  type: sqlite
  data_file_path: /etc/thingsboard-gateway/data/data.db
  messages_ttl_check_in_hours: 1
  messages_ttl_in_days: 2

connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

```

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version: Latest
 - Python version: 3.9

samson0v, 2022-07-12:
Hi @gregwinn, in the 826 issue, there is a problem with the converter. You can use several gateways but you have to change `ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE` parameter for ThingsBoard.
Another solution is to set `minPackSendDelayMS` bigger number, for example, 200 ms.

gregwinn, 2022-07-12:
Thanks, @samson0v I looked into `ACTORS_MAX_CONCURRENT_SESSION_PER_DEVIC` and I want to make sure that would work for me. If I have a single Gateway getting 80 to 150 messages per minute will changing `ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE` do anything? My understanding of that is if I have multiple gateways using the same `accessToken` it will allow more 'gateway instances' to connect as "one".

The issue I am having is when running one Gateway and getting a high number of messages the device is written in things board however none of the telemetry or attributes are written. I also notice it will crash/stop working with no error even with 4 CPU's and 2 Gigs of ram. So I am wondering if the issue is with the Gateway or the amount of data sent to Thingsboard at one? Thank you so much for the help!!

samson0v, 2022-07-13:
Hi @gregwinn, `My understanding of that is if I have multiple gateways using the same accessToken it will allow more 'gateway instances' to connect as "one".` - yes, it is.
Another solution is to set minPackSendDelayMS bigger number, for example, 200 ms.
Also, you can try to enable GRPC for using all your CPU cores.

gregwinn, 2022-07-16:
@samson0v Just an update I am still testing the changes you mentioned. So far, I did enable `grpc` and set `minPackSendDelayMS = 200`.

I did see the same issue, however in only one of the two test gateways I have running.

**NOTE: the two gateways are separate and do not use the same TOKEN.**

The below image is taken from the TB API Usage tab. Please note the drop in messages....

[image]

During the dip in the incoming messages, the logs for BOTH gateways continue to show normal - with no instance of an 'ERROR' / 'CRITICAL' or 'WARNING.'

I checked to see if Thingsboard UI would show me if the Gateway were no longer `active` however, it continued to update hourly with `eventsSent` and `lastActivityTime` showing a current count and date/time.

I was able to grab two errors out of the TB log `/var/log/thingsboard/thingsboard.log` (NOTE: these where the only two errors)
```
2022-07-15 06:57:39,900 [http-nio-0.0.0.0-8080-exec-7] ERROR o.a.c.c.C.[.[.[.[dispatcherServlet] - Servlet.service() for servlet [dispatcherServlet] in context with path [] threw exception
2022-07-15 08:27:56,685 [http-nio-0.0.0.0-8080-exec-11] ERROR o.a.c.c.C.[.[.[.[dispatcherServlet] - Servlet.service() for servlet [dispatcherServlet] in context with path [] threw exception
```

I don't know if they are related to my issue or not but just wanted to add as much context as I could.

### The Gateway that stopped sending telemetry for devices
As you see, the messages did pick back up, but that was only after I restarted the gateway that was down. The only difference I was able to spot between the two gateways was what looks like a memory leak. See the below image.

[image]

### The Gateway that continued to send telemetry
The Gateway that remained "up" has memory that is still going up, just not at the rate as the other gateway. Below is the gateway that continued to work.

[image]

**NOTE: CPU for both remained around 20%, and no spikes.**

I am currently testing the final item by setting `ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE=2` vs. just 1 connection. However, while I have two gateways running, they are NOT using the same TOKEN.

----
Can I provide anything else to help solve this?

gregwinn, 2022-07-17:
Just wanted to add a little more to this. I also found that when the Gateway stops sending telemetry data, the Network TX volume drops, and the memory rises. See below the Gateway that is having issues...
[image]

I thought maybe this particular gateway was doing more traffic, so I compared this to the other Gateway running that has no issues, and you will see below the Network level is the same.
[image]

The above image also makes it clear that memory usage rises rapidly after it stops sending telemetry data.

NOTE: Both Gateways are identical in setup. They are running as containers in docker with the same resources.

samson0v, 2022-07-20:
Hi @gregwinn, we are working on this issue now, so if we will have any results, we will share them with you.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-07-20:
@gregwinn, did you have these two logs (you could missed them because they have INFO and DEBUG status)?
These two logs appear after a while when Gateway gets a lot of messages.
[image]

gregwinn, 2022-07-20:
> @gregwinn, did you have these two logs (you could missed them because they have INFO and DEBUG status)? These two logs appear after a while when Gateway gets a lot of messages and starts to reconnect. [image]

@samson0v yes, I was able to find this in the logs for each Gateway. 

I did notice another pattern, in the logs after the gateway stopped sending telemetry data, I was no longer getting the `Committing changes to DB` & `Sending data to storage` messages. I did continue to get `Successfully converted message from topic /my/topic/name`.

Please let me know if I can provide anything else to help solve this!

gregwinn, 2022-07-20:
@samson0v Below is an image from the logs from a Gateway that is no longer sending telemetry. Please note the absence of `Committing changes to DB `& `Sending data to storage` after letting it run for hours, I have checked the logs and can see where it stops; however, there are no errors/warnings at all.

[image]

gregwinn, 2022-07-20:
@samson0v Progress! After one of the gateways maxed its memory and CPU out I was able to catch this in the logs:
[image]

closed 2022-07-22 by samson0v