thingsboard/thingsboard-gateway #878: How to match key through expressions [HELP]

DaXian459, 2022-07-12:
**For example:"PLC_1A_Q01_Automatic_Status_1"、"PLC_1A_Q01_Automatic_Status_2"、
"PLC_1A_Q02_Automatic_Status_1"。 I want them to match a variable name. Does the gateway support regular expressions**

DaXian459, 2022-07-13:
I use mqtt connection, and I can't find the corresponding description in the document

samson0v, 2022-07-15:
Hi @DaXian459, as you can see from the documentation, MQTT Connector can use these two parameters to parse the device name dynamically.
[image]
[image]

samson0v, 2022-07-18:
Hi @DaXian459, any updates? Can we close the issue?

closed 2022-07-28 by imbeacon