thingsboard/thingsboard-gateway #885: [HELP]

dh-jy, 2022-07-20:
hello , i used   "sudo pip3 install thingsboard-gateway"  to install the gateway, it can work. but i cannot find the log file. 
As shown below.
[image]

samson0v, 2022-07-20:
Hi @dh-jy, follow this [guide](https://thingsboard.io/docs/iot-gateway/install/pip-installation/) to install ThingsBoard IoT Gateway via the pip package.

Thanks for your interest in ThingsBoard IoT Gateway!

dh-jy, 2022-07-20:
yes, i followed it. but in the /var/log/thingsboard-gateway directory i cannot found any log files

发自我的iPhone

dh-jy, 2022-07-20:
thank you for your replay.
i follow the guide to install the gateway. and the gateway can run successfully. but i cannot find the log files in /var/log/thingsboard-gateway 

发自我的iPhone

dh-jy, 2022-07-21:
> @dh-jy, do you run the gateway on the virtual machine?
not , i run the gateway on the Raspberry Pi device

dh-jy, 2022-07-21:
> I have had a similar issue on a gateway - try adding the path and then making sure that the permissions for that folder allow the gateway to write files there. (chmod will let you do this)

hello thank you for your reply, What is this path you are referring to? And How did you install the gateway apt or pip) ?

samson0v, 2022-07-22:
Hi @dh-jy, use the next command:
`sudo chown YOUR_USER:YOUR_USER -R /var/log/thingsboard-gateway`
YOUR_USER - is a user who will run the gateway.

We will update the official documentation in the nearest future.

samson0v, 2022-07-25:
Hi @dh-jy, any updates?

samson0v, 2022-07-28:
@dh-jy, can we close the issue?

dh-jy, 2022-08-01:
> Hi @dh-jy, any updates?

yes, i solve this problem

closed 2022-08-01 by samson0v