thingsboard/thingsboard-gateway #888: [HELP] Connected to TTN and ThingsBoard, but does not create devices

masthdn, 2022-07-25:
**the issue**
I have configured "tb_gateway.yaml" and "mqtt.json". Through the logs, I see I'm connected to both TTN and Thingsboard.
In the logs I can see a successful connection to TTN. Also in Thingsboard, every now and then "lastActivityTime" server attribute of my Gateway device is getting updated.

The issue is that no new device is created when I push a data in TTN. Live data shows my data correctly coming from my sensor node. 

Could you please help?

**Configuration** 
tb_gateway.yaml:
```
thingsboard:
  host:  localhost
  port: 1883
  remoteShell: true
  remoteConfiguration: true
  security:
    accessToken: xxxxxxxxxxxxxxxxxxxxx
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```
mqtt.json:
```
{
  "broker": {
    "name": "Default Local Broker",
    "host": "eu1.cloud.thethings.network",
    "port": 1883,
    "clientId": "ThingsBoard_gateway1", 
    "security": {
      "type": "basic",
      "username": "MY_APP_NAME@ttn",
      "password": "xxxxxxxxxxxxxxxxxxxxxx"
    }
  },
  "mapping": [
    {
      "topicFilter": "+/devices/+/up",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${end_device_ids.dev_eui}",
        "deviceTypeJsonExpression": "LHT65",
        "deviceLabelJsonExpression": "${end_device_ids.device_id}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "application_id",
            "value": "${end_device_ids.application_ids.application_id}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${uplink_message.decoded_payload.TempC_SHT}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${uplink_message.decoded_payload.Hum_SHT}"
[... 178 lines cut ...]
python3[2033]: ''2022-07-25 18:40:46' - INFO - mqtt_connector - 279 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+'
python3[2033]: ''2022-07-25 18:42:16' - DEBUG - mqtt_connector - 314 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f3a7c3381d0>, None, 7)'
python3[2033]: ''2022-07-25 18:42:18' - INFO - mqtt_connector - 237 - MQTT Broker Connector connected to eu1.cloud.thethings.network:1883 - successfully.'
python3[2033]: ''2022-07-25 18:42:18' - DEBUG - mqtt_connector - 243 - Client <paho.mqtt.client.Client object at 0x7f3a7c3381d0>, userdata None, flags {'session present': 0}, extra_params ()'
python3[2033]: ''2022-07-25 18:42:18' - DEBUG - mqtt_connector - 259 - Converter JsonMqttUplinkConverter for topic +/devices/+/up - found!'
python3[2033]: ''2022-07-25 18:42:18' - INFO - mqtt_connector - 279 - Connector "MQTT Broker Connector" subscribe to +/devices/+/up'
python3[2033]: ''2022-07-25 18:42:18' - DEBUG - mqtt_connector - 259 - Converter JsonMqttUplinkConverter for topic /sensor/+/data - found!'
python3[2033]: ''2022-07-25 18:42:18' - INFO - mqtt_connector - 279 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data'
python3[2033]: ''2022-07-25 18:42:18' - DEBUG - mqtt_connector - 259 - Converter CustomMqttUplinkConverter for topic /custom/sensors/+ - found!'
python3[2033]: ''2022-07-25 18:42:18' - INFO - mqtt_connector - 279 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+'
python3[2033]: ''2022-07-25 18:43:42' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:42' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic +/devices/+/up subscription message id = 29'
python3[2033]: ''2022-07-25 18:43:42' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:42' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic /sensor/+/data subscription message id = 30'
python3[2033]: ''2022-07-25 18:43:42' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:42' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic /custom/sensors/+ subscription message id = 31'
python3[2033]: ''2022-07-25 18:43:42' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:42' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic sensor/connect subscription message id = 32'
python3[2033]: ''2022-07-25 18:43:43' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:43' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic sensor/+/connect subscription message id = 33'
python3[2033]: ''2022-07-25 18:43:43' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:43' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic sensor/disconnect subscription message id = 34'
python3[2033]: ''2022-07-25 18:43:43' - INFO - mqtt_connector - 320 - ()'
python3[2033]: ''2022-07-25 18:43:43' - ERROR - mqtt_connector - 325 - "MQTT Broker Connector" subscription failed to topic sensor/+/disconnect subscription message id = 35'
```

masthdn, 2022-07-26:
I'm wondering why I cannot see any of my TTN messages in the log history, while it is successfully connected to.

samson0v, 2022-07-26:
Hi @masthdn, because the MQTT connector can't subscribe to any topics (end of logs).

masthdn, 2022-07-26:
Any suggestion what could have been wrong in my setup?
I'm using a similar setup as suggested here:
https://www.thethingsnetwork.org/forum/t/has-someone-linked-ttn-with-thingsboard/11046/4

samson0v, 2022-07-26:
@masthdn, update your Gateway to the newest version 3.1

masthdn, 2022-07-26:
Ok, I'll try now. Thanks.

masthdn, 2022-07-26:
I updated to the latest version. But still no success. Your help is really appreciated!

Here is my simplified JSON:
```
{
    "broker": {
        "name": "Default Local Broker",
        "host": "eu1.cloud.thethings.network",
        "port": 1883,
        "clientId": "ThingsBoard_gateway1", 
        "security": {
        "type": "basic",
        "username": "cooltag-main-application@ttn",
        "password": "NNSXS.6ZBFBS4RGJO6STX64LBVBFWFYX67EXK5PRPA4UA.RA5FKIESMCPKMWJZF67AFXHUAQOTBF46EEFAK36Q2CR7TERVONGQ"
        }
    },
    "mapping": [
        {
            "topicFilter": "/v3/+/devices/+/up",
            "converter": {
                "type": "json",
                "deviceNameJsonExpression": "${end_device_ids.dev_eui}",
                "deviceTypeJsonExpression": "${end_device_ids.device_id}",
                "timeout": 60000,
                "attributes": [
                    {
                        "type": "string",
                        "key": "application_id",
                        "value": "${end_device_ids.application_ids.application_id}"
                    }
                    
                ],
                "timeseries": [
                    {
                        "type": "double",
                        "key": "temperature",
                        "value": "${uplink_message.decoded_payload.TempC_SHT}"
                    }
                ]
            }
        }     
    ]
 
}
```

Here is the log:
```
Jul 26 04:03:08 ubuntu-2204 systemd[1]: Started ThingsBoard Gateway.
04:03:08" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 112 - Gateway starting..."
04:03:08" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 69 - Checking for new version"
04:03:08" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 117 - ThingsBoard IoT gateway version: 3.1"
04:03:08" - |DEBUG| - [tb_client.py] - tb_client - run - 125 - connecting to ThingsBoard"
04:03:08" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 172 - <paho.mqtt.client.Client object at 0x7f11ea9009d0>"
04:03:08" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 173 - Disconnecting from ThingsBoard"
04:03:08" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 302 - Subscribed to * with id 1"
04:03:08" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
04:03:08" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log service"
04:03:08" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"
04:03:08" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log converter"
04:03:08" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"
04:03:08" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log tb_connection"
04:03:08" - |DEBUG| - [memory_event_storage.py] - memory_event_storage - __init__ - 27 - Memory storage created with following configuration:
Jul 26 04:03:08 ubuntu-2204 python3[6127]: Max size: 100000
Jul 26 04:03:08 ubuntu-2204 python3[6127]:  Read records per time: 100"
04:03:08" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_connector_keys - 1068 - Persistent keys file not found"
04:03:08" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 36 - Root path is: /usr/lib/python3/dist-packages/thingsboard_gateway"
04:03:08" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 38 - Debian installation extensions folder exists."
04:03:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcMqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f11ea900fd0>, origin='/var/lib/>
04:03:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcMqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f11ea901c00>, origin='/usr/lib/>
[... 63 lines cut ...]
04:04:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 246 - Client <paho.mqtt.client.Client object at 0x7f11e9e5c190>, userdata None, flags {'session present': 0}, extra_params ()"
04:04:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter JsonMqttUplinkConverter for topic /v3/+/devices/+/up - found!"
04:04:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /v3/+/devices/+/up"
04:05:17" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 320 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f11e9e5c190>, None, 1)"
04:05:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 241 - MQTT Broker Connector connected to eu1.cloud.thethings.network:1883 - successfully."
04:05:18" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 246 - Client <paho.mqtt.client.Client object at 0x7f11e9e5c190>, userdata None, flags {'session present': 0}, extra_params ()"
04:05:18" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter JsonMqttUplinkConverter for topic /v3/+/devices/+/up - found!"
04:05:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /v3/+/devices/+/up"
[... cut ...]
04:06:24" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 246 - Client <paho.mqtt.client.Client object at 0x7f11e9e5c190>, userdata None, flags {'session present': 0}, extra_params ()"
04:06:24" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_connect - 265 - Converter JsonMqttUplinkConverter for topic /v3/+/devices/+/up - found!"
04:06:24" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 284 - Connector "MQTT Broker Connector" subscribe to /v3/+/devices/+/up"
04:06:43" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 320 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f11e9e5c190>, None, 1)"
```

masthdn, 2022-07-26:
Is there a way to see raw messages to debug my JSON? there might be something wrong in my topicFilter

I use this based on here:
https://www.thethingsnetwork.org/forum/t/thingsboard-ce-connector/52843

samson0v, 2022-07-26:
@masthdn, what version of MQTT you are using?

1 | 0x1 | Connection Refused: Unacceptable protocol version
-- | -- | --

masthdn, 2022-07-26:
> @masthdn, what version of MQTT you are using?
> 
> 1	0x1	Connection Refused: Unacceptable protocol version

Where can I find it?

samson0v, 2022-07-26:
@masthdn, The Things Stack supports the [MQTT Standard Version 3.1.1](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.pdf) and QoS 0 only.

masthdn, 2022-07-26:
> @masthdn, The Things Stack supports the [MQTT Standard Version 3.1.1](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.pdf) and QoS 0 only.

I'm using "ThingsBoard IoT gateway version: 3.1" and also getting connected to TTN community version. Does this answer your question?

samson0v, 2022-07-26:
@masthdn, it was not a question. It means that you have to use MQTT v.3.1.1

masthdn, 2022-07-26:
Where should I install that? 
I followed installation guide from here:
https://thingsboard.io/docs/iot-gateway/getting-started/
and
https://thingsboard.io/docs/iot-gateway/install/deb-installation/

samson0v, 2022-07-26:
@masthdn, you install Gateway in the right way. The problem is that MQTT Broker on eu1.cloud.thethings.network not using MQTT v.3.1.1

samson0v, 2022-07-26:
@masthdn, install paho==1.6.1, that must help

masthdn, 2022-07-26:
> @masthdn, install paho==1.6.1, that must help

It helped a lot already, thanks. I used paho and and could connect to TTN. This is my message:
v3/MY_APP_NAME@ttn/devices/eui-a############/up

Does this show that MQTT Broker on eu1.cloud.thethings.network is v3?

samson0v, 2022-07-26:
@masthdn, the problem was in paho version, not in the MQTT version.

samson0v, 2022-07-28:
Hi @masthdn, any updates? Can we close the issue?

masthdn, 2022-07-28:
I really appreciate all your help. Yes, the issue was solved.

Your suggestion about "paho" helped me find a new path to debug few things. Base on that I found this:
https://github.com/descartes/TheThingsStack-Integration-Starters/tree/main/MQTT-to-Tab-Python3
and then I could found a couple of issues:

**1)**
One was that my topicFilter was
`"topicFilter": "/v3/+/devices/+/up"`
But is should have been this:
`"topicFilter": "v3/+/devices/+/up"`
(note: removing the first slash)

**2)**
I had an network problem. Although I could see TTN (ping and telnet worked fine), the Gateway couldn't talk to. When I used an alternate network, that problem was also solved. For the time being it is enough for my use.

closed 2022-07-28 by masthdn