thingsboard/thingsboard-gateway #892: [HELP] Install thingsboard-gateway v3.1 in Raspberry Pi

alexcorvis84, 2022-08-03:
**Describe the issue**
I'm trying to install thingsboard-gateway v3.1 in Raspberry Pi in following the [deb installation](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)

After issuing `udo apt install ./python3-thingsboard-gateway.deb -y` I'm getting the following message:

[image]

How can I install the missing dependencies?

**Versions (please complete the following information):**
 - OS: [Raspbian GNU/Linux 9 (stretch)]
 - Thingsboard IoT Gateway version [3.1]
 - Python version[3.5.3]

samson0v, 2022-08-03:
Hi @alexcorvis84, I think it can help you:

**Step 1:**

```pip3 install cython
export GRPC_BUILD_WITH_BORING_SSL_ASM=false
export GRPC_PYTHON_BUILD_SYSTEM_OPENSSL=true
export GRPC_PYTHON_BUILD_WITH_CYTHON=true
export GRPC_PYTHON_DISABLE_LIBC_COMPATIBILITY=true
export GRPC_PYTHON_LDFLAGS="-lpthread -Wl,-wrap,memcpy -static-libgcc"
```

**Step 2:**
```
pip3 install --upgrade grpcio-tools --force-reinstall --no-binary=grpcio-tools
pip3 install --upgrade grpcio --force-reinstall --no-binary=grpcio
```
Step 2 took a long time and was sometimes never completed (Raspberry Pi 3). So I did next short cut solution:
Uninstall grpcio;
```
pip uninstall grpcio 
pip uninstall grpcio-status 
```
Then,
```
pip install grpcio==1.44.0 --no-binary=grpcio 
pip install grpcio-tools==1.44.0 --no-binary=grpcio-tools
```

alexcorvis84, 2022-08-03:
> Hi @alexcorvis84, I think it can help you:
> 
> **Step 1:**
> 
> ```
> export GRPC_BUILD_WITH_BORING_SSL_ASM=false
> export GRPC_PYTHON_BUILD_SYSTEM_OPENSSL=true
> export GRPC_PYTHON_BUILD_WITH_CYTHON=true
> export GRPC_PYTHON_DISABLE_LIBC_COMPATIBILITY=true
> export GRPC_PYTHON_LDFLAGS="-lpthread -Wl,-wrap,memcpy -static-libgcc"
> ```
> 
> **Step 2:**
> 
> ```
> pip3 install --upgrade grpcio-tools --force-reinstall --no-binary=grpcio-tools
> pip3 install --upgrade grpcio --force-reinstall --no-binary=grpcio
> ```
> 
> Step 2 took a long time and was sometimes never completed (Raspberry Pi 3). So I did next short cut solution: Uninstall grpcio;
> 
> ```
> pip uninstall grpcio 
> pip uninstall grpcio-status 
> ```
> 
> Then,
> 
> ```
> pip install grpcio==1.44.0 --no-binary=grpcio 
> pip install grpcio-tools==1.44.0 --no-binary=grpcio-tools
> ```

Hi @samson0v as far as I understand these instructions are aimed if you follow the [installation from pip](https://thingsboard.io/docs/iot-gateway/install/pip-installation/). I already found them published by [santobedi user](https://github.com/thingsboard/thingsboard-gateway/issues/709#issuecomment-1125892242).

BUT I'm following the installation as [DEB package](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)

This BUG needs to be addressed somehow and make the installation for the users as easy as it was in the previous versions, without the need to deal with the installation of external dependencies or at least update the documentation accordingly with the official necessary steps. Don't you think?

With the latest versions, is this the normal behavior when installing?

samson0v, 2022-08-03:
@alexcorvis84 I think this instruction can be also useful when you install via the deb package because of system dependencies.
Also, we will remake the deb installation as it was in the previous verions.

samson0v, 2022-08-08:
Hi @alexcorvis84, apt source seems to be outdated and there is no longer this package on the repo for this version of the OS, this usually happens when a new version of the system is released. The fix is either installing with pip or installing the latest version of the OS, which includes the latest repositories, or manually adding a repository with these packages for this version of the OS.

alexcorvis84, 2022-08-08:
> Hi @alexcorvis84, apt source seems to be outdated and there is no longer this package on the repo for this version of the OS, this usually happens when a new version of the system is released. The [...]

I'll try to update the OS packages on device to latest versions and try again the .deb installation. I'll keep you updated with the results

alexcorvis84, 2022-08-08:
Update: I did an apt update & apt upgrade on device and still facing the same unmet dependencies message

imbeacon, 2022-08-08:
@alexcorvis84 ,

This issue doesn't relate to the installation of the gateway, it actually means that there is no package on apt repository for your system, but it is still present in pip, so you can do install via pip using sudo and only after this install the deb package.

alexcorvis84, 2022-08-08:
> @alexcorvis84 ,
> 
> This issue doesn't relate to the installation of the gateway, it actually means that there is no package on apt repository for your system, but it is still present in pip, so you can do install via pip using sudo and only after this install the deb package.

Hi @imbeacon 

I already did (installed missing dependencies via pip), but after issuing again the command to install the .deb package I'm still getting the same message. Please check below image.

[image]

imbeacon, 2022-08-08:
Please try to run `sudo apt -f install ./python3-thingsboard-gateway.deb`

alexcorvis84, 2022-08-08:
> Please try to run `sudo apt -f install ./python3-thingsboard-gateway.deb`

I did and got the same message as before.

imbeacon, 2022-08-08:
Usually it helps with this problem and solves broken packages. You can use the following command as a workaround*:
`sudo dpkg -i --ignore-depends=python3-paho-mqtt python3-thingsboard-gateway.deb` , because you already have installed packages.

alexcorvis84, 2022-08-08:
> Usually it helps with this problem and solves broken packages. You can use the following command as a workaround*: `sudo dpkg -i --ignore-depends=python3-paho-mqtt,python3-grpcio python3-thingsboard-gateway.deb` , because you already have installed packages. Another way is try to use apt with -f -m options.

That way didn't help. It throws even more dependencies are needed.

[image]

imbeacon, 2022-08-08:
Unfortunately it means the same - you cannot get these packages from the repository.
To fix this, you need to change source file for apt by adding the mirror to get the packages.
[How to change the mirror for raspbian](https://pimylifeup.com/raspbian-repository-mirror/)

alexcorvis84, 2022-08-09:
> Unfortunately it means the same - you cannot get these packages from the repository. To fix this, you need to change source file for apt by adding the mirror to get the packages. [How to change the mirror for raspbian](https://pimylifeup.com/raspbian-repository-mirror/)

Thanks @imbeacon for feedback and suggestion. 

I think the shortest way to avoid these problems is upgrading the RaspberryPi OS to a newer version. Currently this device is using the "Stretch" version (which is a bit outdated) and I'm facing these dependencies loop problem, however, on another RaspberryPi I have with "Buster" I don't remember having any problems with dependencies, running .deb installation smoothly. I found this article "How to Upgrade Raspbian Stretch to Raspbian Buster" that might help me.

I wonder if you have tested with RaspberryPi (latest release) and what version of operating system you used?

Thanks in advance.

imbeacon, 2022-08-09:
@alexcorvis84 
You are right, the simplest way is updating the system to the latest version. But if you have some necessary data- you can just change mirrors for the repository.
During testing we are using the latest version of the OS, available at the moment of the testing.

alexcorvis84, 2022-08-09:
> @alexcorvis84 You are right, the simplest way is updating the system to the latest version. But if you have some necessary data- you can just change mirrors for the repository. During testing we are using the latest version of the OS, available at the moment of the testing.

So I'll try to update OS. I'll keep you updated once tested.

samson0v, 2022-08-15:
Hi @alexcorvis84, any updates?

alexcorvis84, 2022-08-16:
> Hi @alexcorvis84, any updates?

Hi @samson0v, I've not been able to perform the test yet. Hope to get an update along this week. I'll keep you posted. Thanks in advance!

samson0v, 2022-08-26:
Hi @alexcorvis84, this bug was fixed in the master branch.
Until the release, you can use Python 3.11.

closed 2022-11-01 by imbeacon