thingsboard/thingsboard-gateway #893: [HELP]

dh-jy, 2022-08-04:
How to increase sample rate in modbus serial (rtu) mode?
as my attampts,change the pollPeriod to 1000ms，500ms， 200ms ，50ms ，the max telemetry recieve rate (i count them in a Timeseries table gadget) is 5 per second (and sometimes 4 or 3 or even 1). how can i increase it?
my configuration file  as the following:
[image]

**Versions (please complete the following information):**
 - OS: [e.g. Raspberry]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2022-08-04:
Hi @dh-jy, send, please, your logs

dh-jy, 2022-08-04:
OK
[image]

dh-jy, 2022-08-04:
No problem uploading data，It just can't achieve high frequency sampling。when i changed the pollPeriod （ Such as：100,50）in the modbus_new_1.json, it cannot imcrease sample.

samson0v, 2022-08-04:
@dh-jy, it seems that 200ms is the maximum poll period for the now.

samson0v, 2022-08-04:
@dh-jy, we will try to increase this value by performing certain optimizations.

dh-jy, 2022-08-04:
OK thank you for your help. 
If you are using sensors to monitor certain devices, high-frequency sampling is important, which means less polling time

samson0v, 2022-08-04:
@dh-jy, managed to get poll period = 60ms

dh-jy, 2022-08-04:
To do this?
[image]

dh-jy, 2022-08-04:
And i  found the code sleep(.2) also influence the poll time， if i set 0.2 to 2s the poll time  will become 2s.
[image]

samson0v, 2022-08-04:
> To do this? [image]

yes

dh-jy, 2022-08-04:
i have set,but it doesn't seem to work.
[image]

samson0v, 2022-08-04:
@dh-jy, you misunderstand me, I said that I made some optimization and manage to poll speed up to 60ms.

dh-jy, 2022-08-04:
ok😂, my bad.could you show the code after optimization

发自我的iPhone

samson0v, 2022-08-04:
@dh-jy made pull request above

closed 2022-08-08 by imbeacon